Asuswrt-Merlin/asuswrt-merlin.legacy #1135: IPTraffic: get hostnames from static DHCP list

zone31, 2016-12-06:
In per device IP traffic, one could pull the device name from the static DHCP list, originData.staticList

[image]

I Have a possible fix, but i have only tested it by replacing the JS functions from the browser console.
pull request coming up :)

zone31, 2016-12-06:
#1136 Possible fix

RMerl, 2016-12-07:
I already retrieve the hostname from the current DHCP lease list.  The reason why I'm reluctant to dig into historical data on top of valid leases or DHCP reservations is because the router has no way of knowing if an IP got allocated to a different client or not over time.  So rather than display inaccurate info (the wrong hostname), I opted to only display the IP, which will always be accurate.

zone31, 2016-12-08:
I see, would it be possible to add this as a checkbox under advanced filter options?

Thynix, 2016-12-30:
It looks like that means #1136 is redundant then? That means static IPs already in the clientList whether connected or not?