Asuswrt-Merlin/asuswrt-merlin.legacy #1139: Enable DNSMasq in AP Mode

mumblecrunch, 2016-12-06:
I would really like to set up the 3 ASUS units in my company's office to provide caching/forwarding DNS services. But they are all in AP mode, so DNSMasq is disabled. Using /jffs/configs/dnsmasq.conf doesn't work when there's no /etc/dnsmasq.conf to override, nor do instructions in /jffs/configs/dnsmasq.conf.add.

I know I can put code in /jffs/scripts/services-start to start dnsmasq with the -C option to point at a config in /jffs/configs/ however as is pointed out in the wiki, this will not survive a services restart. In fact, a services restart is most likely to result either in dnsmasq no longer running or running with no config file.

I understand that it's impossible to provide any kind of reasonable configuration for dnsmasq through the interface, so having the option to enable it with a stub configuration file (effectively disabling both DNS and DHCP by default) that could then be altered or overridden through JFFS would be awesome.

RMerl, 2016-12-07:
In AP mode I wouldn't expect any dnsmasq process launched by the user to get affected (unless Asuswrt blindly looks for a pid regardless of the mode).  In theory, services-start should work for your need.  At worst, you could have a cron job running every few minutes acting like a watchdog, restarting the dnsmasq if it stops running.

mumblecrunch, 2016-12-08:
Per your suggestion, further research has shown that:
• service stop_dnsmasq will stop dnsmasq if it has been started from the start_services script
• service start_dnsmasq has no effect if there's no /etc/dnsmasq.conf file
• Reconfigurations that restart services don't seem to have any effect on a manually started dnsmasq process while in AP mode: I changed the logging configuration, which invoked a services restart, but dnsmasq kept running happily

So while I may put a watchdog in place for a while just for the sake of paranoia, it appears that starting dnsmasq at boot time using services_start and a configuration in /jffs/configs meets my use case. Thanks!

Zulgrib, 2016-12-09:
Couldn't we generate the dnsmasq.conf even when in AP mode ? Wouldn't it be easier ?

closed 2016-12-14 by RMerl