Asuswrt-Merlin/asuswrt-merlin.legacy #1141: Minidlna no longer works when using links

the0ne, 2016-12-08:
After moving to 380.63_2 from some older firmware minidlna didn't serve media any more.
Turning on the logging minidlna finally showed the error:
[2016/12/08 11:01:36] upnphttp.c:1357: error: Rejecting wide link /tmp/mnt/FreeAgent/share/video.mp4 -> /tmp/mnt/FreeAgent/Download2/Complete/folder/video.mp4

By adding
wide_links=yes
to file /jffs/configs/minidlna.conf.add minidlna now works as supposed again.

RMerl, 2016-12-09:
This is for security reasons, to ensure that someone cannot plug a USB disk containing a malicious link that would allow access to security-sensitive files within your router.

Wide links were disabled in Samba a few years ago for the same reason.

This is the intended behaviour.

closed 2016-12-09 by RMerl

the0ne, 2016-12-09:
Agreed. But what about users who already had links set up?

No logs. No warnings. Only when the user want's to play a file, then the client device says 'Media cannot be played back' or the like.

It's really hard to spot what the reason is.
So there could at least be some script checking if there's a link inside a shared folder and add a warning to the minidlna config page in the web-ui.