Asuswrt-Merlin/asuswrt-merlin.legacy #1142: JS files not being cached

bosim, 2016-12-08:
Doing an example request clearly states js files are not being cached. This slows especially HTTPS down. 

    $ curl -v -k https://192.168.1.1:8443/js/jquery.js > /dev/null

    < HTTP/1.0 200 Ok
    < Server: httpd/2.0
    < Date: Thu, 08 Dec 2016 22:55:39 GMT
    < Cache-Control: no-cache
    < Pragma: no-cache
    < Expires: 0
    < Content-Type: text/javascript
    < Connection: close

RMerl, 2016-12-09:
Most JS files contain dynamic content that must absolutely be parsed by the httpd before being sent to clients, therefore they cannot be locally cached.

jquery and chart.js might be the only real exceptions there worth the trouble, and would require to be explicitly declared in the httpd as exceptions.  I might eventually have a look at it, but it's currently not a priority.

RMerl, 2016-12-09:
Since there already was a mime handler defined for jquery, I changed it to enable caching on that file, and also added similar handlers for chart.min.js and ouiDB.js, the largest JS files that are 100% static (commit eced2c3420cbb4c7c9f263884252c93cbaa3692d).

RMerl, 2016-12-09:
Disabling the no-cache header doesn't change anything, browsers are still fully downloading the file every time, requests returning a 200 instead of a 304, load time always remaining at 150-180 ms in my tests.  I suspect the mini httpd daemon used by the firmware might not be advanced enough to truly manage caching with browsers.

bosim, 2016-12-09:
Ok. Thanks for clarifying (really wierd they put dynamic contents in big js files). Would it be possible to optionally enable keep-alive in httpd? I noticed connection: close in js headers.

RMerl, 2016-12-09:
I deleted my last comment, because after looking at the cache content, it seems that there's an expiration date set in 1969, which might explain why nothing gets reused from the cache even though I can see the file in the browser cache.  That might require more in-depth searching.

There's a lot of dynamic content in large files like state.js because they're what make the whole UI work.  For performance reasons, those routers don't have a server-side scripting language such as PHP.  Instead, they dynamically process all sent pages, replacing tags with content generated by the httpd.  The state.js file for instance sets up a lot of variables based on the router current configuration, to be used by the actual pages.

RMerl, 2016-12-09:
Found it (caching must be specifically defined in the handler, done with commit 09800e617f4765084594225bc44879d440e37f80).

Also increased cache time from 5 mins (might be too short for a single webui user session) to 1 hour (commit cb8190683312340b0a0ce6810840dce8f6ddabfd).

RMerl, 2016-12-09:
[image]

closed 2016-12-09 by RMerl

RMerl, 2016-12-09:
The barebone httpd daemon used does not have keep-alive support.  It's a very basic, stripped down server.

bosim, 2016-12-09:
Thanks for looking into it :)