Asuswrt-Merlin/asuswrt-merlin.legacy #1143: SSL certificate field for remote HTTPS administration

Zulgrib, 2016-12-09:
Hello, there's no field to copy/paste an SSL certificate to use instead of the auto generated unsigned one, it would be useful to be able to load our own certificate to be able to check if the channel to communicate with the router was compromised.

If this is already possible, please update the wiki to reflect that.

unsynaps, 2016-12-09:
2 second Google search: https://gist.github.com/davidbalbert/6815258

Zulgrib, 2016-12-10:
"or add it to the wiki"

RMerl, 2017-01-12:
Adding a web interface to manage the webui's SSL certificate is something I'd like to add in the future, however I don't have any ETA as to when that would happen.  Patches are welcome.

RMerl, 2017-05-28:
Commit 7e47adf5b9e7f8785fea9d3dac8597f309b0e70f implements a simpler system than the legacy Tomato method of storing things into an nvram file.  This implementation will store key/certs into JFFS (where OpenVPN already stores its own key/certs), saving the 2-3 KB of nvram space the old Tomato system would have cost.

This also means that users can easily replace them just by copying a new key/cert on top of the existing one under /jffs/ssl/ .  A webui interface would have required the same amount of nvram space, and would have required backend support to be developed and implemented into httpd for moving the files between nvram and the webui, so that plan was dropped - SSH/SCP should be simple enough for users who are advanced enough to generate and provide their own certificate.

closed 2017-05-28 by RMerl