Asuswrt-Merlin/asuswrt-merlin.legacy #1147: Hostfile order

bosim, 2016-12-10:
When I do a traceroute from my LAN I see the following
```
traceroute to google.dk (172.217.20.3), 30 hops max, 60 byte packets
 1  router.asus.com (192.168.1.1)  2.289 ms  2.321 ms  2.432 ms
...
```

My router already has a custom hostname, the hostfile on the router reflects so. It would be nice to get the custom hostname as the first, so dnsmaq would use that for DNS resolution.

RMerl, 2016-12-14:
I will have to make sure it doesn't break anything, as Asus is internally relying on router.asus.com in a couple of places.

unsynaps, 2016-12-17:
Is there anything major dependent on these names? For a while now I have had a script to remove them for the same reason as bosim and have no had any issues (that I can see).

` cat hosts.postconf
#!/bin/sh
CONFIG=$1
source /usr/sbin/helper.sh

pc_replace "192.168.1.1 router.asus.com" "" $CONFIG
pc_replace "192.168.1.1 www.asusnetwork.net" "" $CONFIG
pc_replace "192.168.1.1 www.asusrouter.com" "" $CONFIG`

RMerl, 2016-12-22:
The last two are no longer active (Asus even lost control of these domain names), but first one is used, for instance, by the QIS redirection, probably by the AiProtection block pages as well.  I would not recommend removing it.

commit 0ca3fd48 (2016-12-22): rc: give priority to user-configured hostname rather than the hardcoded Asus ones when generating the hosts file (closes #1147).
closed 2016-12-22 by RMerl