Asuswrt-Merlin/asuswrt-merlin.legacy #1152: Wrongnetwork timezone reported after 380.63_2 upgrade

gena01, 2016-12-16:
So I was running 380.59.0 for a while as my main firmware. A few days ago I upgraded to latest stable. I noticed that a couple of android TV sticks that I have are now showing the wrong timezone (they are set to grab the network date/time settings) and I can't change it easily. :( It also seems that I can not downgrade firmware back to that version (since according to the changelog the file format has changed).

I am in Eastern timezone, but the vmedia android stick and Amazon FireTV stick show both the date/time and GMT+3 Timezone (this shows up wrong in various apps like Kodi and IPTV).

RMerl, 2016-12-16:
The router has nothing to do with the timezone configuration of your device.  In fact, the router doesn't even run an ntpd.

closed 2016-12-16 by RMerl

gena01, 2016-12-16:
Isn't automatic network date/time based on DHCP?

RMerl, 2016-12-16:
No.  A DHCP server can possibly be configured to provide the IP address of an NTP server, but that woudn't affect the client's timezone.  The closest I've been able to find directly tied to DHCP only is some expired RFC that mentions using DHCP option 12 to report the number of seconds since epoch, however that wouldn't include any timezone information either.

What they might be doing potentially is looking at your WAN IP, and base their current timezone on that IP, through geolocation.  If your ISP uses a network block that's used across different regions, it might confuse any geoloation capability however.  For instance my own ISP uses the same IP blocks both in Toronto and Montreal, so some websites incorrectly put me in the wrong city.

gena01, 2016-12-16:
I checked google and a few sites (seems my WAN IP is in the correct timezone or should be). Kinda strange that things have changed recently. Also weird to see both android sticks off by 8 hours and i am unable to fix that.