Asuswrt-Merlin/asuswrt-merlin.legacy #1153: Rounding issue in "Traffic Manager - Traffic Monitor"

Cpuroast, 2016-12-17:
RT-3100
Upgraded to 380.64 final from 380.63_2

[image]

This may be related to https://github.com/RMerl/asuswrt-merlin/commit/168c54266f5987ab3c0289897d8725aace362f2a

RMerl, 2016-12-21:
This looks more like a scaling issue in chart.js, as you have very small values but are using a very large scale.

Cpuroast, 2016-12-22:
Right, after I reverted back to 380.63_2 to test, the same issue occurred.

Once a few days had past with enough traffic, that goes away, so it only happens at the very beginning when the router is first booted.

closed 2016-12-26 by RMerl