Asuswrt-Merlin/asuswrt-merlin.legacy #1155: Ipv6 Not Working on RT-AC3100 380.64

WizardofSloth, 2016-12-19:
Hello, Today I upgraded my RT-N66U to the RT-AC3100. I can't seem to get Ipv6 To work at all. My previous router on the same firmware (380.64) worked fine. I have set it up the same, rebooted and factory restored but nothing seems to get it working. Any suggestions?

zquestz, 2016-12-19:
I can confirm this also broke for me. I am not getting ipv6 connectivity, and even when I ssh to the router a simple `ping6 localhost6` does not work. =\

WizardofSloth, 2016-12-19:
Thank for the reply, I thought I was going crazy! What ISP do you have? I know the whole issue with ISP IPv6 implementation but 380.64 and IPv6 worked fine on my old router. I have Comcast (awful).

zquestz, 2016-12-19:
I am also a comcast user. However my RT-N66U worked just fine to get ipv6 addresses.

zquestz, 2016-12-19:
@WizardofSloth if you ssh into your router, can you ping the ipv6 localhost?

WizardofSloth, 2016-12-19:
@zquestz I got it to start working! I forgot about the setting  TOOLS: OTHER SETTINGS>Firewall: Drop IPv6 neighbor solicitation broadcasts (default: No)	

I switched it to yes, rebooted, and it started working!

zquestz, 2016-12-19:
I am going to give that a shot now, what about local ipv6 pings?

WizardofSloth, 2016-12-19:
ping was successful

zquestz, 2016-12-19:
So strange... I would think `ping6 localhost6` would work whether connected via ipv6 or not, since it is only on localhost. Do you have the ipv6 firewall enabled?

WizardofSloth, 2016-12-19:
I have IPv6 firewall disabled, I have had issues with clients in the past with it on.

closed 2016-12-19 by WizardofSloth