Asuswrt-Merlin/asuswrt-merlin.legacy #1158: OpenVPN UDP Server Regression

zquestz, 2016-12-19:
Just updated to the latest firmware on my RT-AC88U and noticed that my UDP based OpenVPN server no longer functions. I have both the ipv6 and normal firewall enabled, but disabling them does not appear to resolve the issue. My TCP based server continues to work as expected.

I am happy to provide any logs you need to address this.

RMerl, 2016-12-19:
I doubt the issue is related to UDP, because I'm connected at the moment over VPN with my router, using UDP.  No problem accessing my desktop at the remote end of the tunnel using RDesktop.

Recheck the rest of your configuration, especially the custom settings section.  I don't remember seeing any config change in OpenVPN 2.3.14, but they might have changed something else without mentionning it in the changelog.

zquestz, 2016-12-19:
Hmm very odd, I have no custom connection options, and was able to connect fine with both my computer and Android device with the same config on my N66U... Now with my new router, same config, just with the newer firmware it no longer functions...

One other quick question, and perhaps I got a lemon model... I can't even seem to ping ::1 on the router when ssh'd in. Is this expected? I would expect `ping6 ::1` to work, and it doesn't return any packets.

zquestz, 2016-12-19:
Just compared my configs, my UDP and TCP configs are identical except protocol. Arg =(

zquestz, 2016-12-19:
NM, got it working. Needed to change the cipher type. Thanks.

zquestz, 2016-12-19:
Note: exported configs DID change. This was added to my UDP server:

```
cipher AES-128-CBC
```

closed 2016-12-19 by zquestz

RMerl, 2016-12-19:
After a factory default reset, the default cipher is no longer "default" (which for OpenVPN 2.3 means the vulnerable BF cipher) but explicitely AES-128-CBC.  That must be why your exported file changed if you never selected a specific cipher in the past.