Asuswrt-Merlin/asuswrt-merlin.legacy #1159: SSH keys longer than 2048 don't seem to work.

ncrmro, 2016-12-20:
Was wondering if any one has issues with this. I have an ssh key and have manually made key.

unsynaps, 2016-12-20:
This is not all that help full but... Why do you really need SSL keys larger than 2018 on a home router?

ncrmro, 2016-12-20:
Ideally it would support ed25519, but I have trouble using those keys as well on some services Just seems it would be more secure. To super more recent standards. I guess it might be out of the scope of this project.

RMerl, 2016-12-21:
You will have to contact the dropbear author if you want support for new ciphers.  This firmware merely uses his SSH daemon.

A 2048-bit key works fine, I'm using one myself.

ncrmro, 2016-12-21:
@RMerl thanks for pointing me the right direction

closed 2016-12-21 by ncrmro

RMerl, 2016-12-21:
Note that I never tried anything stronger than 2048-bit, but the config buffer should be large enough to support one.