Asuswrt-Merlin/asuswrt-merlin.legacy #1160: Allow IPv6 Client ID Configuration

zquestz, 2016-12-20:
After hours of struggling to get my IPv6 support working with Comcast. I logged into my router and changed the client id passed to odhcp6c. Once I changed the ID, I got an IPv6 connection immediately. Would be really nice to have an option to tune this in the IPv6 setup page.

```
# Broken
odhcp6c -df -R -s /tmp/dhcp6c -N try -c 00030001000000000000 -FP 0:ae57a -r82 -r83 eth0

# Working
odhcp6c -df -R -s /tmp/dhcp6c -N try -c 00030001000000000001 -FP 0:ae57a -r82 -r83 eth0
```

Seems Comcast is ignoring my IPv6 requests on the default client ID, but the moment I change it, everything works as expected.

For now I can work around the issue by just logging into my router after I reboot it, and manually changing the ID, but I am sure other users would find this setting very valuable in the Web UI.

unsynaps, 2016-12-20:
Strange. When I was on Comcast (2 months ago) all I had to do was set it to native... thats it. There is no step 2.

zquestz, 2016-12-20:
@unsynaps this used to work for me as well, but when I upgraded to an RT-AC88U from my old RT-N66U I couldn't get an IPv6 address. I tried rebooting, reflashing, enabling/disabling IPv6 and nothing worked. The moment I changed the client ID I got a lease immediately.

unsynaps, 2016-12-20:
Did you try leaving the modem unplugged for 15 min?

zquestz, 2016-12-20:
@unsynaps I actually only unplugged it for 5 minutes. I didn't do 15. Honesty++. However even with the new ID, it allowed me to get a lease multiple times within seconds...

unsynaps, 2016-12-20:
Yeah the first thing I did when Comcast waws being a toolbag was unplug the mower from the modem and router. Leaving it unplugged for a good 30 min sometimes more (depends on how long it took me to eat dinner when I got home). Then plugged modem in and waited 1-2 min (however long it took to boot) then powered router on. Fixed 99.99% of the problems I never had. Only had to call them 2x in 10 years.

zquestz, 2016-12-20:
@unsynaps Yeah this might be better for the Wiki, just in case people are having issues. Might be that they tie the client ID to a MAC address, and therefore it breaks when you upgrade/change your router. However, instead of waiting 30 minutes+ to get connectivity, this option would make it work immediately, so I think it would be very useful. =)

zquestz, 2016-12-20:
I also did leave IPv6 off all night, then enabled it in the morning, even that didn't seem to help ...

zquestz, 2016-12-30:
Is this something that is feasible? Should I put together a PR?

DwaineGarden, 2017-01-12:
IPv6 broke with latest alpha firmware with Rogers.  It was working fine with previous firmware.  I can confirm that IPv6 is working with Rogers now with the below...

# Working
odhcp6c -df -R -s /tmp/dhcp6c -N try -c 00030001000000000001 -FP 0:ae57a -r82 -r83 eth0

zquestz, 2017-01-12:
Glad that my client id fix was useful for someone else. I looked briefly into doing a PR, need to get back on that.

RMerl, 2017-01-12:
Was it just the ID change that "fixed" it?

zquestz, 2017-01-12:
@RMerl yeah, seems providers sometimes don't like the same ID being used... in my case it was a router upgrade that broke it, and in his it looks like the new firmware, then router reboot caused the ipv6 advertisement to not work. I bet they rate limit that client id in some way. Changing the id causes it to work immediately.

RMerl, 2017-01-12:
Those 000 would indicate the router failed to retrieve your WAN's MAC.  What do you have here?

```
nvram get wan0_hwaddr
```

is it a MAC, or a bunch of 000?

zquestz, 2017-01-12:
AC88U

```
quest@Galactus:/tmp/home/root# nvram get et0macaddr
00:00:00:00:00:00
quest@Galactus:/tmp/home/root# nvram get et1macaddr
00:00:00:00:00:00
```

RMerl, 2017-01-12:
Tracked down a bug in Asus's code where on some router models, it used these incorrect variables to obtain a MAC, leading to these bunches of 000s.  Fixed with c307d95f78b9798d9c764e009adb30e27a7a3a55.

zquestz, 2017-01-12:
@RMerl thanks so much! This is going to help a lot of folks. =)

RMerl, 2017-01-12:
Please retest it again once I publish a new alpha/beta builds (unless you can recompile the FW yourself to test).

I'm not sure it will resolve your specific issue, but it might.

zquestz, 2017-01-12:
I think this should definitely fix it. Any client_id other than 00030001000000000000 works for me when my router refuses to give me a lease. I assume it is because another router is using the same ID. With the MAC fix it should resolve it. I will test when a new build is posted.

thewalkingtoast, 2017-01-12:
The above command fixed it for me with Xfinity SB6141 and AC-3100 running 380.64_2. 👍  Thank you!

thewalkingtoast, 2017-01-13:
After a few hours it reverted and I needed to re-run the command. No router reboot in the meantime.

RMerl, 2017-01-13:
Probably because renewal failed, as it asked once more for MAC 00000000 to obtain a lease.

kavehv, 2017-01-14:
I just built 380.65 beta 1 (8226def) and my RT-AC5300 got an IPv6 address immediately after installation.  Awesome work!

RMerl, 2017-01-16:
Closing this as resolved.

closed 2017-01-16 by RMerl

mike-pt, 2018-01-13:
Sorry to comment on an old one, but I just had to reset my AC66U and before that I was actually cloning the ISP mac address for wan and IPV6 worked perfect.

But the reason I was using that was another (IPTV) and I no longer need to do it  I noticed however that now IPv6 was not working even if the ISP is showing me I have a IPv6 delegated prefix for that mac Address....

So I tried something I killed the running command
`odhcp6c -df -R -s /tmp/dhcp6c -N try -c 00030001d017XXYYZZXX -FP 0:62940 -r23 -r24 -r82 -r83 vlan100`

and simply changed 0003 to 0001 (the rest which is based on the mac was kept the same) and now IPv6 is working....

I should note that I tried disabling/enabling ipv6 countless times, reboot the router and wait (probably no more than 5 mins to be honest)

So my question is where's that fitst part of the ID based of? could we change it in the UI? or better yet if its irrelevant could the code simply attempted to use a different prefix for the ID.... or eeven better maybe it would be possible to have a checkbox for "reset" ipv6 clientID ?

If that is all to much work and you find it pointless I can also live with knowing that I simply have to do this once in a while if its just gets "stuck" not getting the ipv6 address