Asuswrt-Merlin/asuswrt-merlin.legacy #1161: Can't connect to Samba Share

ZhaoXiangXML, 2016-12-21:
之前一直用得挺好的，突然就连不上了，各种客户端都试过。

在后台的管理页面能看到盘和文件夹，迅雷远程下载也还好用，但是共享打不开。

尝试过重启，关掉共享再打开，Format JFFS partition at next boot，均没有效果。还有什么别的可以尝试的吗？

Mac上的提示为：

There are no shares available or you are not allowed to access them on the server. Please contact your system administrator to resolve the problem.

附件为最近一次重启后的log
[syslog.txt](https://github.com/RMerl/asuswrt-merlin/files/666356/syslog.txt)

ZhaoXiangXML, 2016-12-21:
路由器是R6300v2，固件版本是380.63_X7.2

zhangckid, 2016-12-21:
Please discribe your problem in English....Thanks...

mstombs, 2016-12-21:
Which version of Asuswrt-Merlin on which Asus router?
https://twitter.com/x_kayyes/status/670398289439625216

RMerl, 2016-12-21:
He's using someone else's fork, and it's on a Netgear router.

Please contact the author of your firmware for support.

closed 2016-12-21 by RMerl