Asuswrt-Merlin/asuswrt-merlin.legacy #1162: OpenVPN TAP interface doesn't seem to work [380.63_2]

chrisballinger, 2016-12-21:
Everything works fine with TUN configurations, but whenever I try to use TAP in any configuration I get these errors in my Tunnelblick logs. Everything seems to connect and authenticate properly and it says "connected", but there are no routes to any hosts. I would prefer to use TUN, but it doesn't properly handle the Steam multicast packets required for "in-home" streaming.

```
write to TUN/TAP : Input/output error (code=5)
```

I was thinking it was a Tunnelblick/macOS issue, but I set up another OpenVPN TAP server on my network and was able to connect and use the VPN without issue (including remotely via port forwarding). I think the issue is related to not being assigned a DHCP address, and I noticed there are some previous fixes related to that in the changelogs.

Curiously, once I was able to get the two Steam clients to recognize each other via the direct TAP VPN, I can disconnect and reconnect with the router TUN VPN and continue using in-home streaming. I wonder if I can just capture and replay the broadcast packet to bypass the handshake and stick to TUN.

chrisballinger, 2016-12-21:
I'm not sure if this is related, but I also seem to be having problems with "Server 2" in general (which is what I was trying to get to work with TAP). I'm always running a UDP TUN server on "Server 1" and normally also run a fallback TCP TUN on "Server 2". I can no longer get working DNS from Server 2.

I'm wondering if the TAP problems are stemming from "Server 2" being broken in general, but I don't want to mess with my Server 1 configuration at the moment to test this theory.

chrisballinger, 2016-12-31:
I was able to set up a working TAP connection between a Tunnelblick Mac client and an OpenVPN Windows server, so I think the issue is with Merlin's OpenVPN.

RMerl, 2017-01-25:
Going to close this, since there's a number of users using TAP mode without any issue.

closed 2017-01-25 by RMerl