Asuswrt-Merlin/asuswrt-merlin.legacy #1164: Samba permissions don't refresh properly on 2nd partition of USB drive with multiple users

latenitetech, 2016-12-24:
RT-AC68U, AsusWRT-Merlin 380.63_2, "Network Place (Samba) Share / Cloud Disk" screen in the GUI.

I have 2 partitions on a large USB drive, and 2 Samba users (accounts) with different permissions defined.  If I expand all the shares, so all the permissions are shown for all shares, when I toggle between the users (selecting the 'other' user in the left pane), the share permissions update only on the first partition.  To see the true share permissions on the 2nd partition for the 'other' user, I need to refresh the page, then explicitly select the desired user before expanding the share trees.

I don't think it matters to replicate the problem, but here are the details of my specific config:
- 256G USB3 flash drive in USB2 slot.  1 partition formatted FAT32, 2 shares (folders) defined.
- 256G USB3 flash drive in USB3 slot.  2 partitions, the first a ~240G FAT32 w/1 share, the 2nd an 8G EXT2 w/4 shares defined.  
- Both large FAT32 partitions hold media (audio/video files).  The 8G EXT2 partition is for entware and related Linux user/system files.
- 2 Samba users defined:  'admin' has R/W across all shares, 'media' has R-only on the FAT32 shares, no access to the EXT2 shares.

It all works as intended, it's just a bit confusing when reviewing the share permissions and not seeing the proper permissions on the EXT2 partition when toggling between users.  I suspect this is an old bug, and probably not encountered by many as you need both multiple partitions and multiple Samba users defined.

And while I'm at it, can I suggest a feature enhancement?  Add an 'Expand All' button to this screen to expand all share trees.  That's the view I'm always wanting anyway, and it's a bit of a pain to always have to expand each branch manually (for me, 5 extra clicks and the time for the spinning wheel to complete between each one).

Thx for considering this input, and this amazing software effort.

Jemgoss, 2017-05-10:
I experienced the same problem and found the bug in the source. Created a pull request here:
https://github.com/RMerl/asuswrt-merlin/pull/1350

RMerl, 2017-05-10:
Thanks.  I'll review the change, and if it's straightforward enough I will include it for the 380.66 final release, otherwise I'll review it more in depth for the next version.

RMerl, 2017-05-10:
Fixed with commit 0d9d76d4782713ca90397ad0c27ee6308d71db49.

closed 2017-05-10 by RMerl