Asuswrt-Merlin/asuswrt-merlin.legacy #1167: "Get Beta Firmware" Checkbox

Cpuroast, 2016-12-28:
In 380.64, It seems that the "Get Beta Firmware" checkbox has a few problems when using Safari or Chrome.

1. The hit box is off, if I use the tip of the arrow, the checkbox never triggers. The entire arrow has to be on the checkbox for it to trigger.
2. When finally triggered, the state change is temporary, unless "Check" is clicked following the state change, which seems to commit the state change. If however, I immediately hit refresh following the checkbox trigger, it reverts to the opposite state.

These 2 issues do not manifest in Firefox.

Chrome Version 56.0.2924.28 beta (64-bit)
Firefox 51.0b10 (64-bit)
Safari Version 10.0.2 (12602.3.12.0.1)

RMerl, 2016-12-28:
1) Will look into it.  I know there's something wrong with the CSS there as the general visual alignment is wrong, however I never noticed any issue with the click area itself under Chrome.

2) That's a known limitation.  Checking the box cannot send the new state to the server as that page isn't a form with an attached save button like most page (that's why the option to disable scheduled checks had to be put on a different page), not much that can be done without completely redesigning how that page works.

Ninj0r, 2016-12-30:
Hi @RMerl ! I recently got an ASUS router and I really enjoy your firmware.
Thanks for all your hard work!

I think the following might work for you:
```
<span id="beta_firmware_span" style="color:#FFF;display: inline-flex;vertical-align: bottom;">
```

This displays as follows in ```Chrome 55.0.2883.87 (Official Build)```:
[image]

RMerl, 2016-12-30:
Thanks, I'll give it a try.

RMerl, 2017-01-01:
That fixes the alignment, but not the hitbox area itself.

closed 2017-01-01 by RMerl

pepeEL, 2017-01-02:
I think that this function also should support alpha version firmware check. Or add new checkbox to check available alpha version firmware.

RMerl, 2017-01-02:
It already does, however I don't intend to publish alphas to the update server unless they address a serious issue.

commit bd291a8f (2017-01-01): webui: revert f12db7dcd2dd93cde9759d4d5be8a1295655d066 to fix hitbox position on the checkbox (closes #1167); fix installed version display