Asuswrt-Merlin/asuswrt-merlin.legacy #1169: openssl-1.0.0q

umerov1999, 2016-12-29:
This rom has a openssl-1.0.0q and openssl-1.0.2-j. Why the need for openssl-1.0.0q?
(В прошивке присутствуют два openssl - 1.0.0q и 1.0.2-j. Зачем нужна старая версия?)

themiron, 2016-12-29:
because builtin beceem winax support can't work with newer versions. asus legacy

Thynix, 2016-12-30:
Is that something in a firmware blob that we can't fix, or do we have source for that?

RMerl, 2016-12-30:
Closed source, nothing we can do about it.  Probably outside of Asus's control as well, hence they decided to include a legacy version of openssl for those specific components relying on them.

If you don't use a Beceem-compatible 4G modem then this has zero security impact.

closed 2016-12-30 by RMerl