Asuswrt-Merlin/asuswrt-merlin.legacy #1170: Confusing licensing comments

Thynix, 2016-12-30:
The [LICENSE](https://github.com/RMerl/asuswrt-merlin/blob/be6be85e9ec2a4e1c74cd4ac47b2004aa2854cd5/License) indicates this project is GPLv2, but several files, such as [release/src/router/httpd/web.c](https://github.com/RMerl/asuswrt-merlin/blob/2ade8289eea4ea86489b02cc0cc638984ac9b95c/release/src/router/httpd/web.c#L24) and [release/src/router/rc/speedtest.c](https://github.com/RMerl/asuswrt-merlin/blob/master/release/src/router/rc/speedtest.c#L7), contain header comments that "This is UNPUBLISHED PROPRIETARY SOURCE CODE". I see that one of the READMEs [contains](https://github.com/RMerl/asuswrt-merlin/blob/master/release/src/router/README#L7)

```
# THIS SOFTWARE IS OFFERED "AS IS", AND BROADCOM GRANTS NO WARRANTIES OF ANY
# KIND, EXPRESS OR IMPLIED, BY STATUTE, COMMUNICATION OR OTHERWISE. BROADCOM
# SPECIFICALLY DISCLAIMS ANY IMPLIED WARRANTIES OF MERCHANTABILITY, FITNESS
# FOR A SPECIFIC PURPOSE OR NONINFRINGEMENT CONCERNING THIS SOFTWARE.
```

but:

1. Does the GPL apply to the entire project, or not these or other similar files?
2. Does the README's notice supersede the header notices even though it is dated before some of them? If so, would it be valid and make sense to remove the header notices?

RMerl, 2016-12-30:
Those headers were there in the original Asus GPL drop.  Since I didn't write these files, I cannot (and will not) change their headers.

My own work falls under the GPL licence.  Anything else - it's under whatever licence the original authors chose.  And since I'm not a lawyer, I do not start hunting down and over analyzing to determine if those other authors have done anything wrong licence-wise.  You'll have to ask upstream (in this case, Asus) as this came from their own GPL drop.

closed 2016-12-30 by RMerl

Thynix, 2016-12-30:
I see. Thanks for the explanation and prompt response!