Asuswrt-Merlin/asuswrt-merlin.legacy #1171: Add parted tool (currently only fdisk present)

mieszkomularczyk, 2016-12-30:
When I connect 4TB drive to my 68U using the latest merlin firmware (as of Dec 30, 2016) AND the drive is fully cleaned (filled with zeroes), without any partition table, I cannot make use of the drive properly using just the  linux on the router.
The first thing is to create a partition table, however I do need GPT for such a drive. Built-in FDISK also reports the following problem:
_fdisk: device has more than 2^32 sectors, can't use all of them_
I would like to create a single 4TB parition with ext4, however I need full-GPT compatible partition tool. FDISK is old and PC-based Linux systems use PARTED. Please add parted tool to the distribution.

RMerl, 2016-12-31:
Use a Linux live CD.  There's no reason to devote megs of the limited flash to a disk management tool that isn't necessary to run a router, sorry.

closed 2016-12-31 by RMerl

mieszkomularczyk, 2016-12-31:
Hello,
parted takes little space. If so, then I suggest removing fdisk as well. First it is a disk management tool, as you write, second it is useless in the modern days. I would recommend replacing fdisk with parted or if you want to conserve the space so much, remove the fdisk as well, especially if this is useless (cannot create GPT and does not support >2TB drives)

RMerl, 2016-12-31:
fdisk is not a separate tool, it's an applet that's part of Busybox, and therefore only takes a few kilobytes.  I'm leaving it there because it has always been there.

unsynaps, 2016-12-31:
Not to mention fdisk is required to create the partitions the router needs to run **every single time** it starts up (IIRC).

Just write gparted ( http://gparted.org/ ) out to a flash drive and boot off of that to prep the drive before plugging it into the router.

mieszkomularczyk, 2017-01-01:
Hi, thanks for the advice with gparted. Windows' diskpart is not so good, as it always creates 128MB partition for its own purposed if GPT is used