Asuswrt-Merlin/asuswrt-merlin.legacy #1174: WireGuard support

Magissia, 2017-01-01:
Hello, would be nice to have wireguard support

RMerl, 2017-01-01:
Asuswrt-Merlin already provides fairly extensive VPN capabilities, I don't see the need to implement an additional one.

closed 2017-01-01 by RMerl

Magissia, 2017-01-04:
Supposed to consume less resources, not a luxury on our tiny router

mannp, 2017-05-05:
Consumes less resources and be faster than openvpn, which would allow near native speeds on our older routers. Thanks for your excellent firmware. Perhaps its useable via an entware install?

yitzhaq, 2017-10-14:
Merlin,

While I can appreciate not wanting to have to support even more complexity in an already very feature-packed piece of software, I humbly submit that this shouldn't be rejected on the basis that ASUSWRT-Merlin "already supports lots of VPN" (paraphrasing).

WireGuard is a next generation VPN implementation, which is particularly well suited for the types of devices ASUSWRT-Merlin runs on. It lives in kernel, and is far less resource hungry than OpenVPN or similar traditional VPN offerings, resulting in far better throughput, especially on limited hardware. It's built entirely using state-of-the-art crypto in just 4K LoC, and I am not alone in hoping that it will increasingly take over from OpenVPN, IPSec and other unnecessarily complex alternatives.

https://www.mullvad.net/blog/2017/9/27/wireguard-future/

Given its tight nature, it should be possible to compile this into the kernel without sacrificing much precious space, and by the looks of it, this alone should hopefully not require a lot of work. While maintaining UI code for it would no doubt be a more demanding endeavor, just having it available would allow people to play around with it, test it, and write their own custom JFFS scripts for it. If this were to work as well as I would expect it to, UI code could be considered for a later time.

I hope you will reconsider your stance on this, and at least spend a little time investigating the feasibility of simply building ASUSWRT-Merlin with WireGuard support enabled. Hopefully this would not add much maintenance overhead.

Many thanks in advance for your consideration, and even more thanks once again for your continued stellar work on this project.

dopsy, 2017-10-14:
I would also vote for this. Would be great to have this implemented.

yitzhaq, 2017-10-23:
FWIW, testing conducted by VPN provider Mullvad (who offer WireGuard in addition to OpenVPN) showed an increase in throughput from 5-10 Mbps with OpenVPN to 40-60 Mbps with WireGuard on an RT-N66U running LEDE. That's not insignificant.
https://mullvad.net/blog/2017/7/7/how-run-wireguard-router/

lucianconstantin, 2017-11-28:
It might be impossible. [WireGuard requires Linux ≥3.10](https://www.wireguard.com/install/#kernel-requirements) and on my Asuswrt-Merlin-based RT-AC68U router the kernel is 2.6.36.4brcmarm.

unsynaps, 2017-11-28:
Not 100% sure but I think all Asus routers are on that version. Or at least so is mine:
ASUSWRT-Merlin RT-AC87U 380.68-4 Wed Oct  4 19:01:52 UTC 2017
Linux mpu 2.6.36.4brcmarm #2 SMP PREEMPT Wed Oct 4 15:11:10 EDT 2017 armv7l ASUSWRT-Merlin

Not sure what the kernel version in 382.xx is though. My router does not support it yet.

RMerl, 2017-11-29:
Broadcom's new HND platform is the only one with a newer kernel, which means only the RT-AC86U and GT-AC5300.

The kernel is tied to the hardware's SDK, not to the firmware version itself.  I can't be upgraded.

ntam, 2017-11-29:
a good alternative to Openvpn is softether
https://www.softether.org

lucianconstantin, 2017-11-29:
@ntam There's nothing that compares in performance with WireGuard at the moment. Not to mention it's developed by someone with kernel development experience and a lot of security and cryptography experience. It uses modern crypto primitives that are particularly optimized for embedded platforms and it's implemented in less than 4,000 lines of code that can be read in an afternoon (Softether is >300,000 lines of code and OpenVPN is >100,000 and that's without counting OpenSSL). WireGuard's protocol has been formally verified and its performance will only get better once all the possible optimizations that are still on [the TODO list](https://www.wireguard.com/todo/) get implemented. It [has been endorsed by Greg KH](https://plus.google.com/+gregkroahhartman/posts/jD6N4BzToa3), the stable maintainer of the Linux kernel, and it will very likely be integrated directly into the kernel (efforts are underway). It's already in LEDE/OpenWRT and Android custom ROM builders have started integrating it (there will also be a userspace implementation that doesn't require kernel mods). There are commercial VPN providers that have started adding WireGuard servers and at least one offers access for free.

It's just hugely disappointing that SoC manufacturers, especially Broadcom, are holding back progress by not releasing FOSS drivers and forcing devices to use old kernels.

ntam, 2017-11-29:
@lucianconstantin I agree with you in some points but We must face the fact that We can't change the kernel and We must use the tools available and compatible in the tech environment boundary, so I just suggested one like softether which is compatible with almost every SO as well with openvpn clients, furthermore  is secure and get better performance than openvpn, I know that the numbers shows that wireguard can be better than softether, but for the available kernel's limit wireguard is out of scope, another big issue about wireguard stand on the maturity level that can be read on the www.wireguard.com:
"WireGuard is not yet complete. You should not rely on this code. has not undergone proper degrees of security auditing and the protocol is still subject to change. We're working toward a stable 1.0 release, "

having said so, Could be consider softether as alternative? 
regards,

Ubimo, 2018-02-03:
Is wireguard now integrated in Asuswrt?

unsynaps, 2018-02-03:
Uh... no?