Asuswrt-Merlin/asuswrt-merlin.legacy #1176: Use the hardcoded root@asus rather than the real builder's name

HLFH, 2017-01-04:
Hi,

I hate this commit: https://github.com/RMerl/asuswrt-merlin/commit/c89c456a358592ee062d16d8a6c96cf3eaea27c1

On `optware-ng`, `cups` does not work because it needs the `root` user that does not exist, so I can't connect to https://192.168.1.1:631/admin/.

RMerl, 2017-01-04:
This commit has absolutely nothing to do with users, it's only to identify who built the kernel, and identify the code user with the git SHA1.

The "root" user has never existed in Asuswrt, it's always used "admin" as the SUID 1.

closed 2017-01-04 by RMerl

HLFH, 2017-01-04:
@RMerl Ok. thanks for the details. I'll try to switch to Tomato or DD-WRT for the Asus RT-AC3200. I need to get CUPS fully working.