Asuswrt-Merlin/asuswrt-merlin.legacy #1178: WAN Uptime

Cpuroast, 2017-01-04:
Any chance of revisiting this: http://www.snbforums.com/threads/wan-uptime.8570/
Maybe after 380.65 has been released?

Something like john9527's implementation in Tools, below the regular router uptime.

In his implementation, he has "WAN Uptime (Total)" and "WAN Uptime (Current)".

"WAN Uptime (Current)" is probably all that is needed. Would help as an indicator if things are going well with the DSL, Cable or FTTH connection.

Cpuroast, 2017-01-04:
Then again "WAN Uptime (Total)" could be useful too, then you could do a router/wan uptime percentage. So if the WAN is up for less than say 90% of router uptime, there's a problem :)

unsynaps, 2017-01-12:
You would need it to ping some Internet address for it to be accurate. Most cable modems have a internal dhcp server to assign an IP when the coax connection goes down. So just checking if the WAN link is up is not enough.

Cpuroast, 2017-01-12:
Ok, but the cable modem wouldn't assign a public IP if the coax connection went down.

Wouldn't it be possible to detect if there is no IP or a private IP on the WAN and consider that a WAN down scenario?