Asuswrt-Merlin/asuswrt-merlin.legacy #1179: "Error connecting - IP/Routing conflict" error when attempting to setup OpenVPN client

r0bbie, 2017-01-05:
I was drawn to trying out asuswrt-merlin due to its Policy Based Routing capability, in order to direct some traffic through a VPN at the router level while excluding others. I've attempted to set this up, however am continually receving a "Routing conflict" error message when the VPN client is enabled with policy rules in place.

Steps to reproduce:

- Set up for a new VPN client (for instance on client 1). Import .ovpn file, input login information etc, turn on. Test internet and works fine, traffic appears to be being correctly routed through the VPN.

- Turn the VPN client off, and switch the "Redirect Internet traffic" setting to "Policy Rules". Create two rules in order to direct all internet traffic throug the VPN, unless its source is a particular computer at source IP 192.168.1.88 (in line with instructions at https://github.com/RMerl/asuswrt-merlin/wiki/Policy-based-routing): 
```
All traffic	        0.0.0.0	        0.0.0.0	VPN
mediaserver	192.168.1.88	0.0.0.0	WAN	
```

- I get a "Routing conflict" error (screenshots attached below), and the VPN connection seems to not work (no traffic is routed via the VPN).

[image]

[image]

I've tried setting this up multiple times, rebooted the router, etc, and can't see where I'm going wrong. Tried various different policy rules as well to see if I could get any working at all or determine where a conflict might be, with no luck. Had a look through the GitHub issues and also can't see an existing issue which describes this. Assistance appreciated, and let me know if I might provide any log information etc which may be helpful in diagnosing.

RMerl, 2017-01-07:
Please use the support forums at SNBForums for support requests.

closed 2017-01-07 by RMerl

r0bbie, 2017-01-07:
OK thanks - I shall do. I would have done so initially if I hadn't experienced an issue despite setting up rules exactly as described in the documentation, which suggested bug.