Asuswrt-Merlin/asuswrt-merlin.legacy #1181: Upgrade to TOR 0.2.9.8

blackfuel, 2017-01-07:
Requires a simple update to the Asus Makefile to compile TOR.  Just add "--disable-tool-name-check" to the configure line.  For some reason the TOR Makefile did not recognize the ARM cross-compiler.  I have included a patch.

[asuswrt-merlin-380.64_1-tor-disable-tool-name-check.patch.txt](https://github.com/RMerl/asuswrt-merlin/files/691616/asuswrt-merlin-380.64_1-tor-disable-tool-name-check.patch.txt)

**Security fixes since version 0.2.8.9**

Changes in version 0.2.9.8 - 2016-12-19
  Tor 0.2.9.8 is the first stable release of the Tor 0.2.9 series.

  The Tor 0.2.9 series makes mandatory a number of security features
  that were formerly optional. It includes support for a new shared-
  randomness protocol that will form the basis for next generation
  hidden services, includes a single-hop hidden service mode for
  optimizing .onion services that don't actually want to be hidden,
  tries harder not to overload the directory authorities with excessive
  downloads, and supports a better protocol versioning scheme for
  improved compatibility with other implementations of the Tor protocol.

  And of course, there are numerous other bugfixes and improvements.

  This release also includes a fix for a medium-severity issue (bug
  21018 below) where Tor clients could crash when attempting to visit a
  hostile hidden service. Clients are recommended to upgrade as packages
  become available for their systems.

  Below are listed the changes since Tor 0.2.8.11.  For a list of
  changes since 0.2.9.7-rc, see the ChangeLog file.

  o New system requirements:
    - When building with OpenSSL, Tor now requires version 1.0.1 or
      later. OpenSSL 1.0.0 and earlier are no longer supported by the
      OpenSSL team, and should not be used. Closes ticket 20303.
    - Tor now requires Libevent version 2.0.10-stable or later. Older
      versions of Libevent have less efficient backends for several
      platforms, and lack the DNS code that we use for our server-side
      DNS support. This implements ticket 19554.
    - Tor now requires zlib version 1.2 or later, for security,
      efficiency, and (eventually) gzip support. (Back when we started,
      zlib 1.1 and zlib 1.0 were still found in the wild. 1.2 was
      released in 2003. We recommend the latest version.)

  o Deprecated features:
    - A number of DNS-cache-related sub-options for client ports are now
      deprecated for security reasons, and may be removed in a future
      version of Tor. (We believe that client-side DNS caching is a bad
      idea for anonymity, and you should not turn it on.) The options
      are: CacheDNS, CacheIPv4DNS, CacheIPv6DNS, UseDNSCache,
      UseIPv4Cache, and UseIPv6Cache.
    - A number of options are deprecated for security reasons, and may
      be removed in a future version of Tor. The options are:
      AllowDotExit, AllowInvalidNodes, AllowSingleHopCircuits,
      AllowSingleHopExits, ClientDNSRejectInternalAddresses,
      CloseHSClientCircuitsImmediatelyOnTimeout,
      CloseHSServiceRendCircuitsImmediatelyOnTimeout,
      ExcludeSingleHopRelays, FastFirstHopPK, TLSECGroup,
      UseNTorHandshake, and WarnUnsafeSocks.
    - The *ListenAddress options are now deprecated as unnecessary: the
      corresponding *Port options should be used instead. These options
      may someday be removed. The affected options are:
      ControlListenAddress, DNSListenAddress, DirListenAddress,
      NATDListenAddress, ORListenAddress, SocksListenAddress,
      and TransListenAddress.

  o Major bugfixes (parsing, security, new since 0.2.9.7-rc):
    - Fix a bug in parsing that could cause clients to read a single
      byte past the end of an allocated region. This bug could be used
      to cause hardened clients (built with --enable-expensive-hardening)
      to crash if they tried to visit a hostile hidden service. Non-
      hardened clients are only affected depending on the details of
      their platform's memory allocator. Fixes bug 21018; bugfix on
[... 616 lines cut ...]
      potentially wait a very long time. Fixes part of bug 19969; bugfix
      on 0.2.8.1-alpha.

  o Major bugfixes (client performance, backport from 0.2.9.5-alpha):
    - Clients now respond to new application stream requests immediately
      when they arrive, rather than waiting up to one second before
      starting to handle them. Fixes part of bug 19969; bugfix
      on 0.2.8.1-alpha.

  o Minor bugfixes (portability, backport from 0.2.9.6-rc):
    - Work around a bug in the OSX 10.12 SDK that would prevent us from
      successfully targeting earlier versions of OSX. Resolves
      ticket 20235.

  o Minor bugfixes (portability, backport from 0.2.9.5-alpha):
    - Fix implicit conversion warnings under OpenSSL 1.1. Fixes bug
      20551; bugfix on 0.2.1.1-alpha.

  o Minor bugfixes (relay, backport from 0.2.9.5-alpha):
    - Work around a memory leak in OpenSSL 1.1 when encoding public
      keys. Fixes bug 20553; bugfix on 0.0.2pre8.

  o Minor features (geoip):
    - Update geoip and geoip6 to the November 3 2016 Maxmind GeoLite2
      Country database.

RMerl, 2017-01-09:
TransListenAddress and DNSListenAddress are used by Asuswrt, and marked as deprecated by 0.9.  Therefore these should be updated before 0.9 gets merged in, to avoid issues down the road once they do remove the old options.

blackfuel, 2017-01-09:
Here's an updated patch file that also fixes those deprecated torrc options.

[asuswrt-merlin-tor-0.2.9.8-upgrade.patch.txt](https://github.com/RMerl/asuswrt-merlin/files/693482/asuswrt-merlin-tor-0.2.9.8-upgrade.patch.txt)

`cd ~/asuswrt-merlin`
`patch -p2 -i ~/asuswrt-merlin-tor-0.2.9.8-upgrade.patch`

RMerl, 2017-01-12:
Closed by 8033a8d1d7d7c19de9cb4cdee79273b7c815fef9 and ceca9ea33cdcacb6b44076a901b98c8e2ee98409.

closed 2017-01-12 by RMerl