Asuswrt-Merlin/asuswrt-merlin.legacy #1183: Which libusb version?

Nic2012, 2017-01-08:
I am trying to compile on my Asus router RT-AC56U a 3rd party utility to connect IOT devices via usb.
It failed with
`Makefile:305: *** Could not find libusb-1.0 >= 1.0.6.  Stop.`
Which version of libusb-1.0 is used on FW 380.62_1 ?

Afterwards I tried to compile latest version of libusb-1.0.21, but then I got

admin@nas:/jffs/libusb-1.0.21# ./configure
checking for a BSD-compatible install... ./install-sh -c
...
checking for gcc... no
checking for cc... no
checking for cl.exe... no
configure: error: in `/jffs/libusb-1.0.21':
configure: error: no acceptable C compiler found in $PATH

I am not an Linux expert, is a C compiler incl. in FW?
Best regards,
Nic

Nic2012, 2017-01-16:
Installing Entware, recompiling libusb-1.0.21 without libudev support, and some adjustments in 3rd party tool solves this issue. Thanks for non-support ;)

closed 2017-01-16 by Nic2012

RMerl, 2017-01-16:
This isn't a support site, this is a bug tracker.