Asuswrt-Merlin/asuswrt-merlin.legacy #1185: client.ovpn not updating remote with custom DDNS host

kylemart, 2017-01-09:
I just successfully setup a custom DDNS host (i.e. *.duckdns.org) and ddns-start script for my router running 380.64_2. Now I'm trying to setup the VPN server component of the router. When I export the client.ovpn file through the GUI, the line in the file specifying the remote and port contains my router's public IP address, not the custom DDNS host I set. I can manually telnet into the router and edit the line manually to resolve this, but I was wondering if the system should be doing this automatically or if there's a setting through the GUI that I missed.

RMerl, 2017-01-09:
Is the DDNS type set to "Custom", and did you enter your FQDN in the "Host name" field?  Otherwise, the router has no way of knowing what the FQDN would be, so the export process will use the WAN IP instad of a DDNS hostname.

kylemart, 2017-01-09:
Thanks for the quick response! Yes, the DDNS type is set to "Custom" and the host name field is set to, essentially, www.mysubdomain.duckdns.org.

kylemart, 2017-01-12:
http://www.snbforums.com/index.php?threads/35634/

closed 2017-01-12 by kylemart