Asuswrt-Merlin/asuswrt-merlin.legacy #1186: Enhancement Request: Show VPN Status on Left of "App" <Qos> <Wifi> <Guest> <Internet> <USB>

sbollig, 2017-01-09:
If any VPN services are UP then you would see the following:
[image]
If ALL VPN services are DOWN, then you would see this:
[image]
If you wanted to HOVER over the VPN icon you may see what specifically is up: 
[image]
**Server: 1-Connected
Client:  2-Connected**

pepeEL, 2017-01-09:
Good feature +1

pepeEL, 2017-01-24:
RMerlin is any chance to implement this ?