Asuswrt-Merlin/asuswrt-merlin.legacy #1192: CSS layout problem

zuozhehao, 2017-01-12:
[image]
On Mac Chrome.

petertfm, 2017-01-12:
Im also getting this on a rt-ac88u but with Chrome Canary (unstable) on Android

Cpuroast, 2017-01-14:
Yeah, getting the same thing on latest Chrome beta, doesn't happen with Safari or Firefox.
Most pages, the tab bar and the frame are separated by a large empty space

DIdn't happen before, so it could be a Chrome regression or a 364.80_2 regression.

RMerl, 2017-01-14:
Kinda pointless looking at this until Chrome gets out of beta.  You might want to check with their developers first, since the current CSS code is working fine with the latest non-beta browser - it means something they changed is causing this.

Cpuroast, 2017-01-14:
Yeah, that's what I figured :)

RMerl, 2017-01-19:
Closing this.  Based on the screenshot, this wasn't tested on Asuswrt-Merlin but on some other firmware.

closed 2017-01-19 by RMerl

Cpuroast, 2017-01-19:
How is it not Asuswrt-Merlin, it's simply in Chinese. Identical layout as my rt-ac3100 running 364.80_2.

RMerl, 2017-01-20:
They why is the firmware version hidden?  The top showing router model and the Asuswrt-Merlin logo are also truncated, as if the OP wanted to hide something.  My guess is he's running a fork.

Cpuroast, 2017-01-20:
That's entirely possible, it just so happens to have the exact same layout issues as 364.80_2 while using the current beta version of Chrome.

But at this point, since Firefox and Safari have no such issues, I blame Chrome, I'll just see if they eventually fix it before it reaches the final version. If they do not, I suspect more tickets for the same issue will open.