Asuswrt-Merlin/asuswrt-merlin.legacy #1193: Update entware install script to include swapfile creation

latenitetech, 2017-01-12:
The currently installed 'entware-setup.sh' (in /usr/sbin) doesn't include a swapfile creation step, which I think is very beneficial to many users, especially noobies.  The script recommended by this Wiki article:
https://www.hqt.ro/how-to-install-new-generation-entware/
is nearly identical to the current one, except it includes a block of code at the bottom to create a swap file, and also adds code to post-mount to enable it on each boot.

Note that the current built-in script contains code to ensure /jffs is enabled (not included in the external script), so a "best of both" would be the recommended update.

RMerl, 2017-01-19:
@ryzhovau any thought on this?

ryzhovau, 2017-01-20:
I don't think so, sorry. There's a bunch of external services where no swap is required. Most of them.

RMerl, 2017-01-26:
Thanks.  Best to let advanced users do it themselves then, so if they run into stability issues, they can't blame us for it :)

closed 2017-01-26 by RMerl