Asuswrt-Merlin/asuswrt-merlin.legacy #1200: OpenVPN Client policy based routing issue

FrAllard, 2017-01-17:
I setup the OpenVPN client and the connection works perfectly, but no traffic flow through...

I can ping either side endpoint but I cannot ping the remote network.

I added a policy rule as following:
`RemNet123  0.0.0.0  10.77.2.0/24  VPN`

The log file does show this:
`openvpn-routing: Adding route for 0.0.0.0 to 10.77.2.0/24 through VPN client 1`

But the route command from command line does not show the route.

When I manually create the route issuing the following command the tunnel works as expected
`route add -net 10.77.2.0/24 dev tun11`

sbacelic, 2017-01-17:
> Adding route for 0.0.0.0 to 10.77.2.0/24 through VPN client 1

Put 10.77.2.0/24 under Source IP. I guess you got confused by reading [Policy based routing](https://github.com/RMerl/asuswrt-merlin/wiki/Policy-based-routing) where they redirect traffic for all clients (0.0.0.0) to destination IP/range (74.125.0.0/16)

RMerl, 2017-01-24:
Please use the support forums at SNBForums if you need support.

closed 2017-01-24 by RMerl