Asuswrt-Merlin/asuswrt-merlin.legacy #1203: "Plus' button on Advanced_ACL_Content is not working on Chrome

NEVEROYATNII, 2017-01-19:
Was trying to add a MAC addres to a Wireless MAC Filter and after clicking the 'add' button console shows this:
[image]
Select box is not working too:
[image]
Chrome release: 55.0.2883.87 m

RMerl, 2017-01-25:
Did you have any connected wireless client at that time?  The dropdown only contains wifi clients.

NEVEROYATNII, 2017-01-27:
Yea, 1 wireless client. And after i added my device there via Opera it still show nothing in Chrome.
Also, the "client list" on front page isnt working too in chrome.

RMerl, 2017-03-06:
It's possible it might be fixed with 79d5a2709d0aab000ac2c004315684cb30167422 as I'm currently unable to reproduce the issue even by having a test client that contains a single quote in its name.  Check on the Network Map if that's the case among your clients.

NEVEROYATNII, 2017-03-07:
yea, seems like it. Thank you

closed 2017-03-07 by NEVEROYATNII