Asuswrt-Merlin/asuswrt-merlin.legacy #1204: SSH password login possible, regardless disabled in configuration

palokl, 2017-01-19:
Firmware: 380.64_2
HW: RT-AC68U

"Advanced Settings" > Administration > System
  Allow SSH password login : No
  Enable SSH Brute Force Protection : Yes

I have enabled ssh access using SSH Keys which is working well. 
After FW upgrade there is possible also SSH password login (without ssh key) , regardless disabled in configuration.

LOG:
Jan 19 19:36:30 dropbear[30575]: Child connection from x.x.x.x:50817
Jan 19 19:36:37 dropbear[30575]: Password auth succeeded for 'xxxx' from x.x.x.x:50817

RMerl, 2017-01-19:
Please retest using 380.65 Beta 1.  A bug related to dropbear's command line arguments was fixed in it.

palokl, 2017-01-19:
Using 380.65 Beta 1 ssh is working as expected & configured .
Thanks!

Jan 19 20:46:02 dropbear[1495]: Child connection from x.x.x.x:52072
Jan 19 20:46:02 dropbear[1495]: Exit before auth (user 'xxxxxxx', 0 fails): Disconnect received

closed 2017-01-19 by palokl