Asuswrt-Merlin/asuswrt-merlin.legacy #1208: Wifi performance drops

theincogtion, 2017-01-23:
**[1.] One line summary of the problem:**
In an unpredictable manner the Wifi Performance of all devices drop significantly and the latency of locally connected devices increases.
FW:
380.65_beta1
Same issue on stable firmware

Router is running in AP mode. DHCP is done by another device.

**[2.] Full description of the problem/report:**
I can not see the reason nor could I find something in the logs. The Wifi performance drops for all connected devices (android smartphones / windows pcs / raspberry pi) and the latency increases:

```
64 bytes from 192.168.0.108: icmp_seq=1040 ttl=64 time=78.1 ms
64 bytes from 192.168.0.108: icmp_seq=1041 ttl=64 time=297.0 ms
64 bytes from 192.168.0.108: icmp_seq=1042 ttl=64 time=208.9 ms
64 bytes from 192.168.0.108: icmp_seq=1043 ttl=64 time=120.2 ms
64 bytes from 192.168.0.108: icmp_seq=1044 ttl=64 time=31.9 ms
64 bytes from 192.168.0.108: icmp_seq=1045 ttl=64 time=251.1 ms
64 bytes from 192.168.0.108: icmp_seq=1046 ttl=64 time=162.3 ms
64 bytes from 192.168.0.108: icmp_seq=1047 ttl=64 time=77.3 ms
64 bytes from 192.168.0.108: icmp_seq=1048 ttl=64 time=294.9 ms
64 bytes from 192.168.0.108: icmp_seq=1049 ttl=64 time=204.3 ms
64 bytes from 192.168.0.108: icmp_seq=1050 ttl=64 time=116.3 ms
```
In normal conditions I have under 5ms
```64 bytes from 192.168.0.108: icmp_seq=0 ttl=64 time=1.1 ms
64 bytes from 192.168.0.108: icmp_seq=1 ttl=64 time=3.7 ms
64 bytes from 192.168.0.108: icmp_seq=2 ttl=64 time=1.2 ms
64 bytes from 192.168.0.108: icmp_seq=3 ttl=64 time=1.1 ms
64 bytes from 192.168.0.108: icmp_seq=4 ttl=64 time=1.1 ms
64 bytes from 192.168.0.108: icmp_seq=5 ttl=64 time=1.3 ms
64 bytes from 192.168.0.108: icmp_seq=6 ttl=64 time=1.1 ms
64 bytes from 192.168.0.108: icmp_seq=7 ttl=64 time=1.2 ms
64 bytes from 192.168.0.108: icmp_seq=8 ttl=64 time=1.1 ms
64 bytes from 192.168.0.108: icmp_seq=9 ttl=64 time=6.5 ms
64 bytes from 192.168.0.108: icmp_seq=10 ttl=64 time=1.7 ms
64 bytes from 192.168.0.108: icmp_seq=11 ttl=64 time=2.1 ms
```

There are also some strange issues like the problem that I cannot reach my RPi from my PC but from my NAS (both devices are connected to the LAN network)
I can't even ping my Pi from my PC.

Speed from a wifi device if the problem occurs:
138KB/s

Speed after router reboot
5.11MB/s

**[3.] Keywords (i.e., modules, networking, kernel):**

- network 
- wifi
- speed
- latency

**[4.] Kernel information**
**[4.1.] Kernel version (from /proc/version):**
Linux version 2.6.22.19 (merlin@ubuntu-dev) (gcc version 4.2.3) #1 Sun Jan 15 01:33:50 EST 2017
**[4.2.] Kernel .config file:**
couldn't find this file
**[5.] Most recent kernel version which did not have the bug:**
no idea
**[6.] Output of Oops.. message (if applicable) with symbolic information
     resolved (see Documentation/oops-tracing.txt)**

```start_kernel
Linux version 2.6.22.19 (merlin@ubuntu-dev) (gcc version 4.2.3) #1 Sun Jan 15 01:33:50 EST 2017
CPU revision is: 00019749
Found an ST compatible serial flash with 32 64KB blocks; total size 2MB
Determined physical RAM map:
[... 245 lines cut ...]
nf_conntrack_ftp 6976 1 nf_nat_ftp, Live 0xc004f000
wl 3774816 0 - Live 0xc04f6000 (P)
igs 17808 1 wl, Live 0xc006d000 (P)
emf 21152 2 wl,igs, Live 0xc0054000 (P)
bcm57xx 167552 0 - Live 0xc00a3000
et 56320 0 - Live 0xc005e000 (P)```

**[8.4.] Loaded driver and hardware information (/proc/ioports, /proc/iomem)**

```admin@RT-AC66U:/# cat /proc/ioports
admin@RT-AC66U:/# cat /proc/iomem
00000000-07ffefff : System RAM
  00001000-0029c253 : Kernel code
  0029c254-003210bf : Kernel data
18004000-18004fff : ehci_hcd
18009000-18009fff : ohci_hcd
admin@RT-AC66U:/#
```
**[8.5.] PCI information ('lspci -vvv' as root)
[8.6.] SCSI information (from /proc/scsi/scsi)
[8.7.] Other information that might be relevant to the problem
       (please look in /proc and include all information that you
       think to be relevant):
[X.] Other notes, patches, fixes, workarounds:**
My Workaround is to restart the router and everything is fine again.

RMerl, 2017-01-24:
Wifi code is closed source, and outside of my control.

closed 2017-01-24 by RMerl

theincogtion, 2017-01-24:
Any way to troubleshoot this issue or create a bug report for asus?