Asuswrt-Merlin/asuswrt-merlin.legacy #1209: Kernel support for CD-ROM/DVD Filesystems

blackfuel, 2017-01-23:
Here's a use case for adding this extra kernel module to the firmware (CONFIG_ISO9660_FS=m).   I don't think it will break compatibility with the Asuswrt kernel symbols because it's only a module, so it is safe?

https://www.snbforums.com/threads/update-microsoft-windows-and-office-without-an-internet-connection.37020/

RMerl, 2017-01-24:
I've seen cases where even enabling a new module would cause symbol version changes in other components.  That was the case for instance when I was enabling the IPSEC ciphers - it prevented some of the closed source modules to load.

Anything that touches the filesystem subsystem for instance has a risk of affecting the ufsd/tntfs modules, and would need to be thoroughly tested first.

blackfuel, 2017-01-25:
Upon further investigation, after turning on the full support for CD-ROM/DVD filesystems for the RT-AC68U, there are apparently two new kernel symbols that got added.  It did not change any existing symbols, which is good?

**~/asuswrt-merlin/release/src-rt-6.x.4708/linux/linux-2.6.36/Module.symvers**
```
0xf5b4a948	crc_itu_t	lib/crc-itu-t	EXPORT_SYMBOL
0xd29b009f	crc_itu_t_table	lib/crc-itu-t	EXPORT_SYMBOL
```

**How to turn on full support for CD-ROM/DVD filesystems.**  
```
#
# CD-ROM/DVD Filesystems
#
CONFIG_ISO9660_FS=m
CONFIG_JOLIET=y
CONFIG_ZISOFS=y
CONFIG_UDF_FS=m
CONFIG_UDF_NLS=y
```

RMerl, 2017-01-25:
In theory it should, but would have to be tested by ensuring that all third party modules (ufsd on MIPS, tntfs/tfat on ARM, and the Trend Micro DPI engine on ARM) can still properly load without any error in klog/syslog.

RMerl, 2017-05-30:
I won't be adding this module to the firmware, so closing this.

closed 2017-05-30 by RMerl