Asuswrt-Merlin/asuswrt-merlin.legacy #1214: "Error connecting - Authentication failed." while connection is coorectly etablished with OpenVPN

Zulgrib, 2017-01-24:
When using "auth-retry nointeract" option on OpenVPN client, the VPN Status page will say "Error connecting - Authentication failed." if first auth failed, even if the client connected successfully later. The error message doesn't get cleared until manually turning off and on the OpenVPN client.

RMerl, 2017-01-25:
Changing this is going to be difficult, if not impossible.  Whenever an error occurs, OpenVPN triggers an event, and sets the error flag to a value.  However there's no event generated to indicate that a previous error event got cleared (such as a previous auth failure).  That's why Asus only cleared the error flag when the openvpn client is actually started.

So for your specific usage scenario, I can't see any way to work around this.  It would require an extensive study of the OpenVPN code to figure out if there's any way to insert an error clearing call in there - not something I'm willing to devote the multiple hours involved for something that is not an usual usage scenario.

Zulgrib, 2017-01-26:
What should I clear to remove the error message ? I may as well pour a cron that clears it if a ping to a machine on remote lan succeed as a workaround.

RMerl, 2017-01-26:
Set nvram variable "vpn_client1_errno" to 0 (assuming it's the first OpenVPN client, adjust accordingly if not).

You could probably put it into a openvpn-event script instead of a cron job - just need to check if that script gets called after the connection is fully established, or immediately after the client is started (and hasn't connected yet).

(replace client1 with the instance you're using if it's not the first one)

RMerl, 2017-05-30:
Commit 6dcc3cf should take care of this.

closed 2017-05-30 by RMerl