Asuswrt-Merlin/asuswrt-merlin.legacy #1220: icmp type 13 (timestamp request) are allowed in firewall rules, allowing for nmap host discovery on wan ip

schnook, 2017-01-28:
Just noticed that my AC87U was nmap discoverable from wan.  This is due to firewall rules allowing icmp type 13 (timestamp requests).  Home routers should not reply to timestamp requests if it can be avoided, as this allows an attacker to perform host discovery against router wan interfaces.  Recommend release/src/router/rc/firewall.c be examined and changed to fix this default behavior.

**Temporary workaround:**
add the following command to the firewall-start userscript:
`iptables -I INPUT 2 -p icmp -i $1 --icmp-type 13 -j DROP`

Mikiya83, 2017-01-30:
I think it's not a "bug", but a design choice, only icmp type 8 is filtered : https://www.snbforums.com/threads/asus-rt-ac68u-system-log-information-is-someone-accessing-my-network.32068/

schnook, 2017-01-30:
@Mikiya83, thank you for linking me to that thread.  Glad to see this has been thought of before.

Even if it is a design choice, I personally believe it is a poor one, as ICMP type 13 is one of the default host discovery techniques used by the most widely known network security scanning utilities, nmap.  See below from nmap documentation:

> Nmap sends an ICMP echo request, a TCP SYN packet to port 443, a TCP ACK packet to port 80, and an ICMP timestamp request.
[https://nmap.org/book/man-host-discovery.html](https://nmap.org/book/man-host-discovery.html)

On top of that there are many other security concerns with allowing these kinds of scans:

> Receiving an ICMP Timestamp Reply would reveal an alive host (or a networking device) that has
implemented the ICMP Timestamp messages.
[http://www.blackhat.com/presentations/bh-europe-00/OfirArkin/OfirArkin2.pdf](http://www.blackhat.com/presentations/bh-europe-00/OfirArkin/OfirArkin2.pdf)

This article, from the SANS institue, talks about using type 13's to fingerprint the server you are sending them to:
[https://www.sans.org/security-resources/idfaq/how-can-attacker-use-icmp-for-reconnaissance/3/13](https://www.sans.org/security-resources/idfaq/how-can-attacker-use-icmp-for-reconnaissance/3/13)

I will rest my case after this post, but I do want to raise the issue.  Heck, if we don't care about timestamp requests, why bother blocking echo requests either?  I am willing to bet that the person seeing these requests in @Mikiya83's linked thread was being hit by an automated scanner, akin to those running scans against services like telnet.  Do we really want to willingly provide a potential attacker _any_ information about our network, as small as it may be?

At least merlin's build allows for custom firewall rules...

Mikiya83, 2017-01-31:
I agree with you and add your rule, but i'm interested in Merlin's point of view of this filtering, maybe theses messages are needed for an "hidden" task in router's services, so it's important to be sure before addind this permanently.

RMerl, 2017-02-01:
I'm always reluctant in arbitrarily blocking specific types of traffic, as you never know when some weird ISP or application might be broken by such a change.  At least with IPv6, the RFCs are very clear as to which ICMP types should be allowed in, but I don't think such an authoritative reference exists for IPv6.

What is the typical behaviour of other firewalls/router software out there?

schnook, 2017-02-01:
It looks like there is an RFC that defines the necessity of icmp type 13/14.  From page 43 of RFC 1122:

> A host MAY implement Timestamp and Timestamp Reply.

So it looks like timestamp requests/replies are **optional**.  Again, I would recommend we block these in the INPUT chain.

In addition, I ran an nping scan for timestamp requests against my pfSense box, and it looks like pfSense is configured to NOT reply to timestamp requests by default.

As far as whether some weird ISP relies on this type of network traffic, I have no idea, but I doubt they would be relying on it for anything.  Same goes for applications; no clue if there are weird ones that will break without timestamp replies,

RMerl, 2017-02-01:
Might be good to compile a list of ICMP types that should/could/must be blocked rather than just put a special case for ICMP 13 (with echo being left untouched, as it's already controllable).  Those additional types could then be controlled through a separate nvram setting  (exposed to the webui or not), in case someone's specific application might need any of these to be allowed in.

Would be interesting to look at the INPUT chain of other projects such as OpenWRT/DD-WRT to see what's the norm in what's allowed in and what isn't.

schnook, 2017-02-01:
@RMerl Very true.  The blackhat article I linked above listed these protocols as all the ones that are ICMP "queries":

> ECHO Request (Type 8), and Reply (Type 0)
Time Stamp Request (Type 13), and Reply (Type 14)
Information Request (Type 15), and Reply (Type 16)
Address Mask Request (Type 17), and Reply (Type 18)
 Router Solicitation (Type 10), and Router Advertisement (Type 9)

It looks like the router doesn't implement most of these ICMP types by default (alot of them are deprecated), which is why I did not mention them.

I just fired up an old wrt54g with a shibby build of tomato, and it looks like they handle the firewall rules by setting the default INPUT chain policy to DROP.  In the case of the shibby firmware, it looks like nothing is explicitly allowed.  Keep in mind though, that most ICMP types necessary for normal internet communication will be covered by the stateful "ALLOW all RELATED, ESTABLISHED" rules.  With that in mind, it kind of makes me wonder why ASUS even decided to include a rule explicitly allowing all other types of icmp.  Whatever...

RMerl, 2017-02-02:
I took a look at Tomato.  It has an option to allow/block WAN PING (like Asus), however the code incorrectly will allow or block all ICMP types.  That means allowing pings will allow any other ICMP types as well.  Tomato is not a good reference I guess.

RMerl, 2017-02-02:
wl500g: they allow all ICMP packets in, except for echo (which will depend on whether you allow them in or not).

RMerl, 2017-03-06:
@themiron I'm curious as to what's your opinion on this one?  I'm tempted to drop 13,15 and 17 out of the box.  I suppose 10 could be dropped too, since it's only used by ICMPv6 (which has a separate, RFC-compliant list of allowed/dropped types).

schnook, 2017-03-08:
@RMerl To be honest, after thinking more about the rules Asus has decided to implement, I honestly think if we just remove this rule it should take care of everything:

`    0     0 ACCEPT     icmp --  *      *       0.0.0.0/0            0.0.0.0/0            icmp !type 8`

Sure, we will be denying many ICMP types,  but all the important ones should be covered by the RELATED, ESTABLISHED directives.  Reference this article for more information:
[http://www.linuxtopia.org/Linux_Firewall_iptables/x1571.html](http://www.linuxtopia.org/Linux_Firewall_iptables/x1571.html)

RMerl, 2017-03-08:
RELATED/ESTABLISHED relates to stateful connections such as tcp, they shouldn't have any effect on ICMP.

schnook, 2017-03-08:
@RMerl RELATED/ESTABLISHED do apply to ICMP traffic as well, to a certain extent.  As per this article:
[http://www.iptables.info/en/connection-state.html](http://www.iptables.info/en/connection-state.html)

Relevant bits:

> ICMP reply messages can also be considered as ESTABLISHED, if we created a packet that in turn generated the reply ICMP message.
>[The RELATED state] could be used to allow ICMP error messages, FTP transfers and DCC's to work properly through the firewall.

In addition to the two articles I referenced, I have also tested these rules myself and can confirm that ICMP information does in fact get handled by RELATED/ESTABLISHED rules.

RMerl, 2017-03-15:
It would handle a reply, but still wouldn't take care of dropping undesirable types such as icmp 13.

themiron, 2017-05-07:
@RMerl
Timestamp  Request (13) can be safely dropped like and when ICMP Echo request does.
Information Request (15), Address Mask Request (17) are no-op in 2.6.36.
IPv4 Router * (9, 10) are not implemented in 2.6.36 at all.

So, I think, there's 2 options:
1. add additional WEB UI option to control that, unlikely you want it :)
2. just control ICMP Timestamp with ICMP Echo setup due Timestamp attack delongs to host discovery type as well as ICMP Echo

Also, I think it's not worth to block ICMP Timestamp early before accepting RELATED/ESTABLISHED like it's done for ICMP Echo, because early blocking bloats INPUT chain only to stop replying to frequent ICMP Echo messages coming in less than ICMP conntrack timeout == 30s.  ICMP Timestamp messages are not expected to be sent frequently due they are used for host one-time discovery mostly, so it's quite safe to block only new in absence of established ICMP Timestamp stream.

So.. please refer
https://github.com/themiron/asuswrt-merlin/commit/83a147b305bcffcb9394617f845204cd30890afe

RMerl, 2017-05-10:
I like the idea of moving ICMP to its own chain if we're gonna start filtering multiple packets.  That idea should probably be extended to the long list of icmpv6 types that are accepted in ip6tables as well.

RMerl, 2017-05-28:
This change has been implemented by Asus with GPL 380_7627.

closed 2017-05-28 by RMerl