Asuswrt-Merlin/asuswrt-merlin.legacy #1224: curl and wget not able to connect to ipv6 page.

gelsbern, 2017-01-30:
I am writing a custom ddns start script and need to update both the ipv4 and ipv6 address.  The urls look like this (authentication information changed)

curl -k -4 "https://mydomain.com:password@dyn.dns.he.net/nic/update?hostname=mydomain.com" 
curl -k -6 "https://mydomain.com:password@dyn.dns.he.net/nic/update?hostname=mydomain.com"

The ipv4 one works great, however ipv6 always times out.  I have tested the curl lines from a linux machine on the same network ant it works there just fine.  I cannot understand why the router itself cannot connect to that ipv6.

unsynaps, 2017-01-30:
1: HE TB needs to be able to ping your router
2: There is something broken with HE's 6in4 on the router. With it enabled it prevents SAMBA from starting and causes a ton of other problems.

gelsbern, 2017-01-30:
I am not using  HE's tunnel as my ISP has native IPV6  however I am using their free dns server with the dynamic dns. 

Per their instructions here: https://dns.he.net/docs.html
All I am wanting to do is to write the ddns-start custom script.
I am running the latest version.   As stated curl -4 works fine, but curl -6 doesn't  And curl -6 works on everything inside the lan, so I know that the router is doing ipv6 properly.

unsynaps, 2017-01-30:
Use no.ip or something that already has DDNS support built into the router?

gelsbern, 2017-01-30:
I really appreciate the help of what you are trying to do.  I use dns.he.net for all my domains.  I have for years.  Up until recently the domain I have this router for was using the HE tunnelbroker, but as I said my ISP recently went native, so I can skip the tunnel but I still want to use the ddns part.  I am not going to switch nameservers.  There is an issue, I am reporting it, I am not going to bypass it or change by DNS provider or anything, I am trying to use a  documented feature:  https://github.com/RMerl/asuswrt-merlin/wiki/Custom-DDNS

So I appreciate what you are trying to do, with your suggestions, it is not really helping.  Hopefully the developer will take a look at this and perhaps fix the issue so that the documented feature can be fully used.

gelsbern, 2017-01-30:
As a further update, curl and wget aren't able to get to ANY ipv6 site using those utilities from the router itself.  This includes google.com  so it's not restricted to he.net.

RMerl, 2017-01-30:
Working for me using an HE tunnel:

```
admin@Stargate88:/tmp/home/root# curl -6 https://ipv6.google.com -v > test.txt
* Rebuilt URL to: https://ipv6.google.com/
*   Trying 2607:f8b0:400b:80a::200e...
* TCP_NODELAY set
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* Connected to ipv6.google.com (2607:f8b0:400b:80a::200e) port 443 (#0)
* ALPN, offering http/1.1
* Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
* successfully set certificate verify locations:
*   CAfile: /rom/ca-bundle.crt
  CApath: none
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
} [5 bytes data]
* TLSv1.2 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
* TLSv1.2 (IN), TLS handshake, Server hello (2):
{ [106 bytes data]
* TLSv1.2 (IN), TLS handshake, Certificate (11):
{ [3757 bytes data]
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
{ [148 bytes data]
* TLSv1.2 (IN), TLS handshake, Server finished (14):
{ [4 bytes data]
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
} [70 bytes data]
* TLSv1.2 (OUT), TLS change cipher, Client hello (1):
} [1 bytes data]
* TLSv1.2 (OUT), TLS handshake, Finished (20):
} [16 bytes data]
* TLSv1.2 (IN), TLS change cipher, Client hello (1):
{ [1 bytes data]
* TLSv1.2 (IN), TLS handshake, Finished (20):
{ [16 bytes data]
* SSL connection using TLSv1.2 / ECDHE-ECDSA-AES128-GCM-SHA256
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: C=US; ST=California; L=Mountain View; O=Google Inc; CN=*.google.com
*  start date: Jan 18 19:17:59 2017 GMT
*  expire date: Apr 12 18:51:00 2017 GMT
*  subjectAltName: host "ipv6.google.com" matched cert's "*.google.com"
*  issuer: C=US; O=Google Inc; CN=Google Internet Authority G2
*  SSL certificate verify ok.
} [5 bytes data]
> GET / HTTP/1.1
> Host: ipv6.google.com
> User-Agent: curl/7.51.0
> Accept: */*
> 
{ [5 bytes data]
< HTTP/1.1 200 OK
< Date: Mon, 30 Jan 2017 15:04:53 GMT
< Expires: -1
< Cache-Control: private, max-age=0
< Content-Type: text/html; charset=ISO-8859-1
< P3P: CP="This is not a P3P policy! See https://www.google.com/support/accounts/answer/151657?hl=en for more info."
< Server: gws
< X-XSS-Protection: 1; mode=block
< X-Frame-Options: SAMEORIGIN
< Set-Cookie: NID=95=YNzAIHn3LyqQoY7M2tz1PmgjsLfVvdB39nqfWZS8VOImHKJvI8uFs39aWmzm3xthRIpeJjDJu9MoNqYySQIk7laGT6nubspHcZpYHYBkLXSDHe1UUvk9AJU1iKyHMaXu; expires=Tue, 01-Aug-2017 15:04:53 GMT; path=/; domain=.google.com; HttpOnly
< Alt-Svc: quic=":443"; ma=2592000; v="35,34"
< Accept-Ranges: none
< Vary: Accept-Encoding
< Transfer-Encoding: chunked
< 
{ [496 bytes data]
* Curl_http_done: called premature == 0
100 10255    0 10255    0     0  55487      0 --:--:-- --:--:-- --:--:-- 56038
* Connection #0 to host ipv6.google.com left intact
```

gelsbern, 2017-01-30:
Hrmm, I am not using the tunnel, I am using Native and this is what happens:

> gelsbern@Ecclesianet:/tmp/home/root# curl -6 http://ipv6.google.com -v
> * Rebuilt URL to: http://ipv6.google.com/
> *   Trying 2607:f8b0:4009:80e::200e...
> * TCP_NODELAY set
> * connect to 2607:f8b0:4009:80e::200e port 80 failed: Connection timed out
> * Failed to connect to ipv6.google.com port 80: Connection timed out
> * Closing connection 0
> curl: (7) Failed to connect to ipv6.google.com port 80: Connection timed out

Also, just so you can see that ipv6 is working:

> gelsbern@Ecclesianet:/tmp/home/root# ping -6 ipv6.google.com
> PING ipv6.google.com (2607:f8b0:4009:80e::200e): 56 data bytes
> 64 bytes from 2607:f8b0:4009:80e::200e: seq=0 ttl=53 time=27.238 ms
> 64 bytes from 2607:f8b0:4009:80e::200e: seq=1 ttl=53 time=28.092 ms
> 64 bytes from 2607:f8b0:4009:80e::200e: seq=2 ttl=53 time=28.229 ms
> 64 bytes from 2607:f8b0:4009:80e::200e: seq=3 ttl=53 time=27.617 ms
> 64 bytes from 2607:f8b0:4009:80e::200e: seq=4 ttl=53 time=27.651 ms
> 64 bytes from 2607:f8b0:4009:80e::200e: seq=5 ttl=53 time=27.062 ms
> 64 bytes from 2607:f8b0:4009:80e::200e: seq=6 ttl=53 time=27.598 ms
> 64 bytes from 2607:f8b0:4009:80e::200e: seq=7 ttl=53 time=26.718 ms
> 64 bytes from 2607:f8b0:4009:80e::200e: seq=8 ttl=53 time=27.545 ms
> ^C
> --- ipv6.google.com ping statistics ---
> 9 packets transmitted, 9 packets received, 0% packet loss
> round-trip min/avg/max = 26.718/27.527/28.229 ms
>

gelsbern, 2017-01-30:
I have updated to the beta3 version and it does the same thing.

gelsbern, 2017-02-02:
I am now at beta4, issue still persists.  Is there any thing I can get you as far as debugging Native IPV6.  Both curl and wget are unable to connect via ipv6 from the router's shell.

gelsbern, 2017-02-14:
Just wanted to mention I am now at the latest release.  Seems something up with native ipv6.  Can't figure it out. Tunnel works fine because it goes out over ipv4.

RMerl, 2017-03-06:
Someone else will have to look into this, as I have no way of testing native IPv6 support.

shounak-de, 2017-04-01:
FYI: I have native ipv6 and `curl -6 http://ipv6.google.com -v` works for me. 
My curl is from entware-ng, router is mips (rtac66u), firmware is 380.65

RMerl, 2017-09-09:
Closing this as it's working correctly for me as well as others.

closed 2017-09-09 by RMerl