Asuswrt-Merlin/asuswrt-merlin.legacy #1233: DHCP forwarding/passthrough not working

fonix232, 2017-02-04:
As mentioned in [my thread](https://www.snbforums.com/threads/problem-with-repeater-mode.37299/), I've tested some versions of asuswrt-merlin, going back all the way to 380.62 (sans the beta updates, excluding 380.65_beta4).

The result is quite clear. Asuswrt-merlin introduced a change, that, compared to Asus' official 380_4161, will not allow DHCP passthrough or forwarding to work when the router is in Repeater Mode.

I made the following configuration:
- Reset router completely (nvram erase)
- Upon boot, connect wired, and select repeater mode
- Select the host AP (5GHz channel, hidden SSID, gateway is 192.168.10.10, netmask is 192.168.10.0/24)
- Set fixed IP (192.168.10.50, with DNS set to Google's)
- Set up custom AP name, with password
- Let it connect, and renew DHCP-received address

In case of the two official Asus firmwares, DHCP address is received immediately after connecting to the router either wired or wireless.

In case of the tested asuswrt-merlin firmwares, neither the router nor the host router will provide DHCP service, thus the device receives no IP, and cannot access the network. Setting the IP address manually will resolve this issue, and provide network level connection - setting the host router as gateway will provide Internet connection too.

Unfortunately I found no mention of the method used for DHCP forwarding, so I could not try any manual methods.

DranZoch, 2017-02-06:
Great someone else with the same problem, my clients  receives sporadically the wrong IP-configuration, wrong subnet and gateway. the gateway is 0.0.0.0 instead of the main router, which yield no internet/no network. Also it is loosing the IP after 24h. I could improve it by increasing the DHCP lease time and static IP for the repeater. but as soon as I restart the clients they have no access. After I restart the repeater it works sometimes. And as fonix232, setting the IP/subnet/gateway manually works.

The repeater is connected to the main router with the 5Ghz radio (WPA2) and the clients use the 2.4 Ghz (WPA2). the wired seems much more stable but there is still sometimes no network access.

I will be thankful if you have time to look into this.

fonix232, 2017-02-06:
In my case clients connected through the Asus AP do not receive DHCP addresses at all.

steeve, 2017-03-19:
Same here, but only when uplink is 5ghz

shanepcs, 2017-04-24:
I think I am also experiencing this issue.

KarlRoby, 2017-12-02:
I'm experiencing a similar issue.

When I add a Windows Server DHCP server to the network and disable the DHCP server function of my Asus RT-AC88U the DHCP requests are not forwarded to the Windows Server properly.

Wired clients obtain leases correctly, but it seems to take about 60 seconds to get a lease. Wireless clients don't seem to get a lease. Sometimes an iPhone will get a lease, but not consistently. And anything through a repeater or bridge never gets a lease, even if I release the lease before disabling the router's DHCP server and enabling the Windows on.