Asuswrt-Merlin/asuswrt-merlin.legacy #1243: OpenVPN error after the last upgrade ( Firmware:380.65)

GustaFo0, 2017-02-12:
After each upgrade I go to the open VPN tab to set up my account. 
As usual I download the latest *.opvn files from HMA server...

Then this:

**System Log - General Log**

Feb 11 18:41:10 openvpn[31832]: OpenVPN 2.4.0 arm-unknown-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] built on Feb  3 2017
Feb 11 18:41:10 openvpn[31832]: library versions: OpenSSL 1.0.2k  26 Jan 2017, LZO 2.08
Feb 11 18:41:10 openvpn[31833]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Feb 11 18:41:10 openvpn[31833]: TCP/UDP: Preserving recently used remote address: [AF_INET]107.190.164.130:553
Feb 11 18:41:10 openvpn[31833]: Socket Buffers: R=[122880->122880] S=[122880->122880]
Feb 11 18:41:10 openvpn[31833]: UDP link local: (not bound)
Feb 11 18:41:10 openvpn[31833]: UDP link remote: [AF_INET]107.190.164.130:553
Feb 11 18:41:10 openvpn[31833]: TLS: Initial packet from [AF_INET]107.190.164.130:553, sid=e10e03a4 7f078b8e
Feb 11 18:41:10 openvpn[31833]: WARNING: this configuration may cache passwords in memory -- use the auth-nocache option to prevent this
Feb 11 18:41:15 openvpn[31833]: VERIFY OK: depth=1, C=UK, ST=London, L=London, O=Privax Ltd, OU=HMA Pro VPN, CN=hidemyass.com, emailAddress=info@privax.com
Feb 11 18:41:15 openvpn[31833]: VERIFY OK: nsCertType=SERVER
Feb 11 18:41:15 openvpn[31833]: VERIFY OK: depth=0, C=UK, ST=London, L=London, O=Privax Ltd, OU=HMA Pro VPN, CN=server, emailAddress=info@privax.com
Feb 11 18:41:18 openvpn[31833]: WARNING: 'link-mtu' is used inconsistently, local='link-mtu 1558', remote='link-mtu 1557'
Feb 11 18:41:18 openvpn[31833]: WARNING: 'comp-lzo' is present in local config but missing in remote config, local='comp-lzo'
Feb 11 18:41:18 openvpn[31833]: Control Channel: TLSv1, cipher TLSv1/SSLv3 DHE-RSA-AES256-SHA, 2048 bit RSA
Feb 11 18:41:18 openvpn[31833]: [server] Peer Connection Initiated with [AF_INET]107.190.164.130:553
Feb 11 18:41:19 openvpn[31833]: SENT CONTROL [server]: 'PUSH_REQUEST' (status=1)
Feb 11 18:41:20 openvpn[31833]: PUSH: Received control message: 'PUSH_REPLY,dhcp-option DNS 10.200.137.1,ping 9,ping-restart 30,route-gateway 10.200.137.1,topology subnet,redirect-gateway def1,ifconfig-ipv6 2001:db8:123::2/64 2001:db8:123::1,route-ipv6 2000::/3 2001:db8:123::1,explicit-exit-notify 2,ifconfig 10.200.137.225 255.255.255.0'
Feb 11 18:41:20 openvpn[31833]: Option 'explicit-exit-notify' in [PUSH-OPTIONS]:9 is ignored by previous <connection> blocks 
Feb 11 18:41:20 openvpn[31833]: OPTIONS IMPORT: timers and/or timeouts modified
Feb 11 18:41:20 openvpn[31833]: OPTIONS IMPORT: explicit notify parm(s) modified
Feb 11 18:41:20 openvpn[31833]: OPTIONS IMPORT: --ifconfig/up options modified
Feb 11 18:41:20 openvpn[31833]: OPTIONS IMPORT: route options modified
Feb 11 18:41:20 openvpn[31833]: OPTIONS IMPORT: route-related options modified
Feb 11 18:41:20 openvpn[31833]: OPTIONS IMPORT: --ip-win32 and/or --dhcp-option options modified
Feb 11 18:41:20 openvpn[31833]: Data Channel Encrypt: Cipher 'AES-256-CBC' initialized with 256 bit key
Feb 11 18:41:20 openvpn[31833]: Data Channel Encrypt: Using 160 bit message hash 'SHA1' for HMAC authentication
Feb 11 18:41:20 openvpn[31833]: Data Channel Decrypt: Cipher 'AES-256-CBC' initialized with 256 bit key
Feb 11 18:41:20 openvpn[31833]: Data Channel Decrypt: Using 160 bit message hash 'SHA1' for HMAC authentication
Feb 11 18:41:20 openvpn[31833]: GDG6: remote_host_ipv6=n/a
Feb 11 18:41:20 openvpn[31833]: TUN/TAP device tun11 opened
Feb 11 18:41:20 openvpn[31833]: TUN/TAP TX queue length set to 100
Feb 11 18:41:20 openvpn[31833]: do_ifconfig, tt->did_ifconfig_ipv6_setup=1
Feb 11 18:41:20 openvpn[31833]: /usr/sbin/ip link set dev tun11 up mtu 1500
Feb 11 18:41:20 openvpn[31833]: /usr/sbin/ip addr add dev tun11 10.200.137.225/24 broadcast 10.200.137.255
Feb 11 18:41:20 openvpn[31833]: /usr/sbin/ip -6 addr add 2001:db8:123::2/64 dev tun11
Feb 11 18:41:20 openvpn[31833]: Linux ip -6 addr add failed: external program exited with error status: 2
Feb 11 18:41:20 openvpn[31833]: Exiting due to fatal error

______________________________________________________________________________________

I do not know enough to figure out what is wrong??? 
But I think it is something to do with the latest OpenVPN 2.4.0

Any ideas??

Thanks

Syst3mSh0ck, 2017-02-13:
Feb 11 18:41:20 openvpn[31833]: Linux ip -6 addr add failed: external program exited with error status: 2

error status 2 = Operation not permitted

Further information on this error from OpenVPN Wiki: https://community.openvpn.net/openvpn/wiki/UnprivilegedUser

This is a privilege escalation issue with initialization script. Hope this helps.

RMerl, 2017-02-13:
The fix has been posted on the support forums.  See the first post:

https://www.snbforums.com/threads/release-asuswrt-merlin-380-65-is-now-available.37295/

closed 2017-02-13 by RMerl