Asuswrt-Merlin/asuswrt-merlin.legacy #1244: Issues/FYI

JakeDEvans, 2017-02-13:
Not exactly an issue, but wanted others to know.

If you add IPv6 to your openvpn tunnels, you will need to additionally add rules to your IPv6 firewall if you want traffic to traverse.

I'm not sure if this is intentional or not.

VPN Expert options for IPv6 Tunnels (replace subnets with your vpn and lan subnets, I used (2) from my /48 HE.net tunnel

```
push "comp-lzo yes"
tun-ipv6
server-ipv6 2001:db8:200::/64
push "route-ipv6 2001:db8:100::/64"
push "route-ipv6 2001:db8:200::/64"

```

RMerl, 2017-02-19:
IPv6 is not supported by OpenVPN tunnels.

closed 2017-02-19 by RMerl

JakeDEvans, 2017-02-19:
Bullshit it isn't.

RMerl, 2017-02-19:
Let me rephrase that: the OpenVPN implementation in Asuswrt-Merlin does not support IPv6.  If you want to use IPv6, you have to configure everything yourself.  This is not a usage scenario that I support.

There's no need to be rude.

JakeDEvans, 2017-02-19:
rude begets rude. it works, all I'm saying.