Asuswrt-Merlin/asuswrt-merlin.legacy #1252: bug - wireless scheduler not working after using the wireless button

dilighiani, 2017-02-22:
I have set a wireless scheduler to stop the WiFi signal every night after 1:00 AM. Sometimes I need to have the WiFi signal open after 1:00 AM and I force to enable the WiFi signal from the front button of the AC87U. But after that, for the following days, the scheduler is disabled until the momment when I logon to the web Interface to re-enable the WiFi schedule.
I consider this an Asus WRT bug but I don't know how to open bug requests for Asus WRT.

hggomes, 2017-02-22:
You just simply need to send an email to: networking_support@asus.com

dilighiani, 2017-02-23:
Thank you for the tip hggomes. I have sent an email to them.

hggomes, 2017-02-24:
You are welcome, thank you also for reporting it.

closed 2017-03-01 by RMerl