Asuswrt-Merlin/asuswrt-merlin.legacy #1253: OpenVPN Client not persisted on reboot?

stevomccormack, 2017-02-22:
My OpenVPN Clients are not persisted on reboot?
Is this a feature??
Are you able to provide a solution?
Is there a command I can put into JJFS that will run and reconnect VPN on startup?
---

Furthermore, I have been on support to Express VPN for hours on end over many days and my VPN speed is 90% reduction. I get 120MB download when not using VPN and then 20MB maximum when using OpenVPN client. Express VPN are adamant it is not them. Telstra is a possible cause (throttling) but from all research I have read this would be against the law in Australia possibly and no evidence exists.

So my final conclusions are the OpenVPN .ovpn settings are perhaps incompatible with ASUSWRT Merlin or the OpenSSL is not well supported on router so encryption process is bottle neck. 

Thanks.

p3yot3, 2017-02-27:
I too have this problem. When I restart/reboot my PC's/Servers etc I also have to restart my router VPN clients in order to reconnect, even though I have persist-tun & persist-key in the conf.

UK based using PIA.

RMerl, 2017-03-01:
You need to enable the "Start with Wan" option.

closed 2017-03-01 by RMerl

p3yot3, 2017-03-01:
It already is. That's the problem.