Asuswrt-Merlin/asuswrt-merlin.legacy #1255: Bug with 380.65 and mediaserver.asp

seattlecajun, 2017-02-23:
I updated the firmware on my ASUS RT-68U this morning from an earlier version of AsusMerlin to 380.65 and have since been having a funky issue on the mediaserver.asp page.

I enabled the Media Server and tested dlna, all was working perfectly as expected.  Then, I tried configuring Manual Media Server paths.  I setup a few and hit save.  Now, whenever I view the Media Server page, I get this:

[image]

I'm unable to move the page upward or downwards and in order to access any other part of the UI, I have to reload and go back to my main page's address (192.168.11.2).  It doesn't matter which browser or PC/Mobile device that I view the page on, the behavior is the same.

I reset my device back to factory defaults and set everything up again.  After that, it worked perfectly until I tried setting up the manual media server paths again, same issue.

Any ideas what could be going on?

rkkoszewski, 2017-03-01:
Hi, I have the same router with the same firmware and I'm using manual paths. Everything is working fine here. 

Do you have any special characters in your folders or drive name? Maybe thats interfering with the XML reaponse generated in the admin panel?

seattlecajun, 2017-03-03:
Hey there, I'm glad to hear the issue affects only me and other folks aren't having trouble.  It's definitely pretty odd behavior.  I ended up flashing down, as it were, to the Asus branded firmware to test that out.  It's working as expected under that firmware.  

I may try to go back to Asusmerlin's build here soon and give it another go soon but I've just been too busy lately. As for folder names, some of them are a bit on the longish side but they don't contain any special characters.  It's a good direction to dig deeper though.

RMerl, 2017-03-06:
Open your browser console and look for any Javascript error.  Chances are your device contains an invalid character such as a quote, breaking the webui code.

Syst3mSh0ck, 2017-03-07:
Shouldn't there be a sanity check in place to prevent this kind of breakage from occurring to begin with?

seattlecajun, 2017-03-13:
Thanks for getting back to me RMerl! I was never able to track the issue down, unfortunately.  I ended up plugging the drive into my PC and reconfiguring the folder structures, upgraded back to the latest merlin build and all was well again. :)

closed 2017-03-13 by seattlecajun