Asuswrt-Merlin/asuswrt-merlin.legacy #1256: jffs has many strange files

ctgreybeard, 2017-02-23:
My /jffs has started accumulating many files with names that start with 'nmp' followed by apparently random nonsense characters.

For instance:
```
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmp
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmp
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmp
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmp
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmpϤ
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmpд
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmpж
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmp
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmp
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmp
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmp
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmpӒ
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmp
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmp
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmpԩ7
-rw-rw-rw-  384 admin    root           840 Feb 23 09:31 nmp
```

These appeared around the time that the AC66 spontaneously rebooted but I have seen them before and cleared them out.

It appears that the names are really of the form: `nmp�9�ent_list`

Does that mean anything to someone? Some sort of client list?

I was able to print out the contents of one of the files:

```
<784f437baba9>>ruusan>>0>1>0>0<24c86e04538f>>*>>0>1>0>0<a82066165a60>>TATOOINE>>0>1>0>0<4494fc7fb99d>>>>0>1>0>0<00626e526210>>>>0>1>0>0<b827ebcbcdda>>scanmon>>0>1>0>0<b827eb568c4f>>nagios>>0>1>0>0<a0f3c1c45e2f>>>>0>0>0>0<f48139d51213>>000000D51213>>0>1>1>0<b827eb813ff0>>camera1>>0>0>0>0<b827ebd46aa5>>camera1>>0>0>0>0<c869cd660fcf>>Waggoners-TV>>0>0>0>0<b8f6b1135fef>>Grundtvig>>0>0>0>0<080027eab85b>>hesperos>>0>1>0>0<080027b9ae96>>jakku>>0>1>0>0<18b4304d1cdd>>*>>0>0>0>0<645299224c21>>MyQ-1D7>>0>1>0>0<c82a14018c5e>>trantor>>0>0>0>0<b8e85697eb1e>>BillWagnersiPad>>0>0>0>0<9c04ebaaa759>>Karens-iPad>>0>0>0>0<d0034b868a8b>>BillWagsiPhone2>>0>0>0>0<78fd9422a5e3>>KarenWarsiPhone>>0>0>0>0<18b4301e8992>>*>>0>0>0>0<cc95d78d5d21>>*>>0>0>0>0<c0cecde70e6f>>BillWsApleWatch>>0>0>0>0<e0f84739bb02>>trantor>>0>0>0>0<0874024de70b>>Tricorder>>0>0>0>0admin@RT-AC66R-571
```

Those are the names of several of the systems on my network.

fonix232, 2017-02-28:
It looks like a binary dump of the connected devices.

Do the names "trantor" "TATOOINE", "Waggoners-TV", "Grundtvig", "hesperos", "jakku", "BillWagnersiPad" etc. sound familiar?

ctgreybeard, 2017-03-01:
Oh yes, those are machines on my network. I cleared them all out and now there are 122 new ones. Most dating from Feb 24.

closed 2017-03-01 by RMerl