Asuswrt-Merlin/asuswrt-merlin.legacy #1261: RT-AC56U 380.65 - Port Forwarding broken

chrysikitty, 2017-02-28:
After upgrading to the latest firmware, 380.65, I noticed port forwarding appears to be completely broken. I've tried multiple times adding and deleting and turning on and off and several reboots and nothing works. I even added the port forwarding manually using an SSH connection and editing IPtables directly and it still wouldn't forward the ports. Since I'm running a public facing server from behind this firewall I can't wait for a patch and will have to try a different firmware for now but thought I'd let you know.

Sorry for not attaching any logs. I'm not really sure which ones would be useful.

chrysikitty, 2017-02-28:
I think I fixed it. I had to do a factory restore and reboot. But it seems to be working now.

closed 2017-02-28 by chrysikitty

larsosca, 2017-02-28:
Hi, Are you sure?
I have a similar problem and I did the same.
It worked first but failed after a while, sometimes stopped working after some minutes and other times it worked up to an hour.