Asuswrt-Merlin/asuswrt-merlin.legacy #1264: NAT Loopback is not work

cutedogspark, 2017-03-02:
Dear RMerl
      I updated the firmware 380.65 version on my ASUS RT-68U, when I  Factory default Route then the NAT Loopback is not work, pls help me

bossarts, 2017-04-27:
Hello

Same issue here, see config:
ASUS RT-68U
380.65_2 -> Install with factory reset
380.66 beta -> Update
NAT Loopback: Merlin/Ausus -> both are not working
OpenVpn Server installed
192.168.223.1 Router IP
DHCP 192.168.223.200-230
SUBNET 255.255.255.0
STATIC DHCP for Webcams
WEBCAM1: 192.168.223.30
WEBCAM2: 192.168.223.32

Port Forwarding List
WEBCAM1	80 port <--> WAN 33860 port
WEBCAM2	80 port <--> WAN 33862 port

Testing:
LAN	-> myddns.ch:33860 fail
LAN -> myddns.ch:33862 fail
4G(WAN)	-> myddns.ch:33860 success
4G(WAN) -> myddns.ch:33862 success
LAN -> WANIP:33860 sucess
LAN -> WANIP:33862 sucess
Conclusion: NAT loopback works when WAN IP is used but not with DDNS myddns.ch

DNS Reserch on Win10 CMD:
C:\> nslookup myddns.ch
Server: Routername.myddns.ch
Address: 192.168.223.1 -> This is the Router IP which seems OK
***superboss.me wurde von Routername.myddns.ch nicht gefunden: Non-existent domain.

I'm lost. Any ideas?

Regards

RMerl, 2017-04-27:
Your problem is your DDNS, not the NAT loopback.  Look at the error message on your nslookup result.

NAT loopback is working fine in the firmware, lots of people are using it without any issue.

closed 2017-04-27 by RMerl

bossarts, 2017-04-27:
Changed "RT-AC68U's Domain Name"  to someting else than myddns.ch and now it works! Thanx for the hint!