Asuswrt-Merlin/asuswrt-merlin.legacy #1265: Update nano to 2.7.5

feranick, 2017-03-02:
v. 2.7.5 is now available as a bug fix.

GNU nano 2.7.5 "Nijntje" can properly search and replace the \B and \b regex anchors, correctly repaints things when multiline regexes with identical start and end are involved, fixes a crash with zero-length regex matches, does replacements at the edges of a marked region right, 
no longer hides double-width characters at the head of softwrapped rows, displays at most three warnings at startup, and documents the ability to read a file from standard input. 

https://www.nano-editor.org/news.php

feranick, 2017-04-03:
Just checking if this is going to be included in the current git. Thanks.

RMerl, 2017-04-03:
Probably not, since I don't expect a new firmware release for the coming weeks and this nano release is pretty minor, I will probably wait for the next one once I get closer to a new release - unless they're still at 2.7.5 by then.

feranick, 2017-04-03:
Makes sense, Thanks.

feranick, 2017-04-04:
New Release is out, v.2.8.0. Changelog:

GNU nano 2.8.0 "Axat" makes it easier to move around in softwrapped lines: the Up and Down keys now step from visual row to visual row instead of jumping between logical lines, and the Home and End keys now move to the start and end of a row, and only when already there, then to the start and end of the logical line. Furthermore, the screen can now scroll per row instead of always per logical line.  On an entirely different front: nano now makes use of gnulib, to make it build
on more platforms.  In short: there were many internal changes, not many user-visible ones (apart form the new softwrap navigation).

closed 2017-04-10 by RMerl

feranick, 2017-04-14:
Hi, 2.8.1 has been released as a bug fix... changelog below. I would normally open a new bug report, but given that you are still in beta phase, I just drop this in here, knowing that it is probably a little late for merging it into 380.66... 

GNU nano 2.8.1 "Ellert" fixes build failures on MacOS and on musl, fixes scrolling problems in softwrap mode when double-width characters on row boundaries are involved, shows double-width characters as ">" and "<" when split across two rows, moves the cursor more predictably (at
the cost of sometimes putting it on the second "half" of a character), avoids creating lines that consist of only blanks when using autoindent, makes ^Home and ^End go to the start and end of the file (on terminals that support those keystrokes), places the cursor better when linting, lets the linter ask only once whether to open an included file, and adds bindings for ^Up and ^Down in the file browser.

feranick, 2017-05-05:
Hi, 2.8.2 is out... Probably too late for this development cycle... but here it is if you feel like it can still be added.

GNU nano 2.8.2 "Krats" adds another new feature: it makes the ^G help texts searchable with ^W.  Apart from that, it fixes a crash when resizing the window in the middle of verbatim input, avoids an unlikely crash when used without UTF-8 support in some locales, avoids redrawing the screen twice when switching between buffers while line numbers are active, and works around a coloring
bug on musl.  Plus tweaks to the documentation; plus translation updates for fifteen languages.

feranick, 2017-05-11:
Just checking if still worth bumping this up to 2.8.2 in 380_66. Thanks.

RMerl, 2017-05-11:
Too late for that now, and there's nothing important enough in that update.

commit ad72afd9 (2017-04-10): nano: Updated to 2.8.0 (closes #1265)