Asuswrt-Merlin/asuswrt-merlin.legacy #1267: SSH not working for WAN access

ntam, 2017-03-02:
Hi,

I enable the SSH for WAN&LAN access, test it and was working as expected, after that I changed the port to 2222, and stop working even if I set back to the default port value to 22, I tried reset to factory and also reebooitng after applying any changes.

Router: RT-AC68u
Merlin Version: 380.65

let me know if you need any extra data like logs or whatever needed to find the what cause the issue.

Regards,

ntam, 2017-03-11:
I discover that is the Firewall that is blocking the access to ssh for wan, I disable the firewall and the ssh wan was granted, I turn on the firewal and again I cant access to ssh service for wan.

unsynaps, 2017-04-02:
It really is not a good idea to open the web or ssh to the WAN interface at all.

skogsmaskin, 2017-08-08:
@unsynaps - why not SSH? If it secured properly with a private key and password login is disabled, it should be safe.  You hare correct about the web interface though.