Asuswrt-Merlin/asuswrt-merlin.legacy #1268: Start with WAN not working on 380.65

p3yot3, 2017-03-02:
Even though the Start with WAN setting is enabled I have to restart the VPN client manually in order to connect every time I switch on or reboot my computers. There is talk of others with the same issue on the SNB forum:

https://www.snbforums.com/threads/how-to-setup-a-vpn-client-including-policy-rules-for-pia-and-other-vpn-providers-10-15-fixed.30851/page-16

N66U Router using 3 clients.

Syst3mSh0ck, 2017-03-07:
I had the same issue with PIA VPN on RT-AC56U using several different firmware versions. After increasing the log verbosity I could see the tunnel was dropping randomly every 4 to 26 hours due to time-out even with echo disabled. WAN connection is rock solid. The tunnel would re-establish several times but after a while it would stop sending or receiving any frames, necessitating a reboot of the VPN client. I believe the root cause of the issue to be with the VPN provider, or an upstream bug with OpenVPN itself rather than Merlins firmware.

p3yot3, 2017-03-07:
I've never had this issue with PIA or any other firmware. If I revert back to the previous version, everything works fine again with the same settings or slightly different ones (I've tried a few variations from different guides in an effort to stop the disconnects when not in use).

p3yot3, 2017-03-15:
I've updated to 380.65_2 but still have the same problem. Tried factory reset, etc but the problem persists.

p3yot3, 2017-04-02:
I've updated to 380.65_4 but still have the same problem.

p3yot3, 2017-04-09:
many users are reporting the same issue on the snb forums:

https://www.snbforums.com/threads/how-to-setup-a-vpn-client-including-policy-rules-for-pia-and-other-vpn-providers-10-15-fixed.30851/page-17

Can the dev here please have a look into the problem - having to constantly restart my vpn clients is getting a little infuriating.

Thank you.

amgarrety, 2017-04-09:
On which version was it last reliably working for you?  I just upgraded from the stock firmware to merlin specifically for this feature and I can't get it working on 380.65_2 or 380.65_4.

RMerl, 2017-04-11:
Start with WAN is working for me.  Your log doesn't indicate any issue with that, but with an issue in reconnecting once your connection times out, which is a totally different issue.

RMerl, 2017-04-11:
I also tested with a tunnel configured with a TLS renegotiation set to 60 secs (so I don't need to wait for an hour to reproduce the soft resets you see in your log), and it also worked properly for me using VPNBook - I was still able to go through the tunnel after the TLS was renegotiated.

I suspect your issue lies with a config parameter specific to PIA (so far the only reports I've seen were also with PIA).

closed 2017-04-11 by RMerl

p3yot3, 2017-04-11:
I took the advice of others on the snb forums & reverted back to 380.64_2 & since doing so I've not had a single disconnect either in use or after rebooting using PIA & the same settings, so there is definately something amiss with the firmwares after 380.64_2 & PIA.
There is talk on the SNB forums of the latest alpha working fine with PIA, so I guess I'll just wait for the official next release & keep using the 380.64_2 firmware until then.

Thanks anyway for your firmware & the work you do.

RMerl, 2017-04-11:
The latest alpha updated OpenVPN to 2.4.1, so it's possible that it was a bug in 2.4.0.