Asuswrt-Merlin/asuswrt-merlin.legacy #1270: nslookup returns IPv6 addresses first even if IPv6 is disabled

raducotescu, 2017-03-03:
Running on 380.65.

```bash
# nslookup google.com
Server:    127.0.0.1
Address 1: 127.0.0.1 localhost.localdomain

Name:      google.com
Address 1: 2a00:1450:400a:807::200e zrh04s08-in-x0e.1e100.net
Address 2: 172.217.18.110 zrh04s05-in-f14.1e100.net
# uname -a
Linux scotty 2.6.36.4brcmarm #2 SMP PREEMPT Fri Feb 3 00:27:49 EST 2017 armv7l ASUSWRT-Merlin
```

[image]

This leads to Segmentation Fault for both `wget` and `curl`.

RMerl, 2017-03-06:
It's normal for a resolver to return both A and AAA records on lookups, regardless of whether the environment supports IPv6 or not, as a resolver has no way of knowing if the requesting client has a working IPv6 route.  It's up to clients to handle it properly.  Here when I try it with wget here, it correctly fails with a "network is unreachable", and it retries again with an IPv4 afterward, which succeeds.

closed 2017-03-06 by RMerl

raducotescu, 2017-03-07:
Ok, I can accept this resolution. However, do you have any idea why both `curl` and `wget` fail with a segfault?

raducotescu, 2017-03-20:
@RMerl, do you have any suggestion how I could get rid of the mentioned segfault?